Ultimaker/Cura #19372: Conan fails to install one of the packages

pskowronskiTDx, 2024-07-12:
Here is my current configuration.
 
- Visual Studio Professional 2022 - 17.10.3
- CMake 3.26.3
- Ninja 1.11.1
- Conan 1.60.2
- sip 6.8.3
- python 3.10.7

Conan installation fails as follows (I keep Cura main branch up to date):
[image]

Full conan install log:
[conan_log.txt](https://github.com/user-attachments/files/16157507/conan_log.txt)

_Originally posted by @pskowronskiTDx in https://github.com/Ultimaker/Cura/issues/19295#issuecomment-2219828235_

wawanbreton, 2024-07-12:
I see two suggestions in the conan error message, the first one is not easy to setup, but did you try to set `CONAN_DISABLE_STRICT_MODE=1` before building ?

pskowronskiTDx, 2024-07-15:
Ok, thank you, I must have missed that. Setting mentioned variable did the job. Nevertheless, having the installation completed I keep running into pyArcus issue when trying to launch Cura.

pskowronskiTDx, 2024-07-15:
The application cannot launch at all (I am still referring to #19299). I am not going to work on the engine so it would be nice if there is a way to work around pyArcus.

wawanbreton, 2024-07-16:
Well you could remove the calls to pyArcus, they are located in two files and there are not hundreds of them, most being only for typing.

pskowronskiTDx, 2024-07-16:
Ok, I have managed to remove references to the modules which are using pyArcus without making the application crashing. I hope you will find a way to resolve this DLL issue in the future.

wawanbreton, 2024-07-16:
I'm glad to know that you can finally start Cura !
The DLL issue comes from some incompatibility between binaries produced with different versions of the MS compiler. Not sure there is something we can do :grimacing: we may at some point get rid or arcus though.
I was also thinking that maybe there is a way to force Conan to rebuild all the binaries, that may help.