Ultimaker/Cura #19373: Add "Enable / Disable" USB Printing menu option to Extensions menu

GregValiant, 2024-07-12:
### Is your feature request related to a problem?

There have been a number of complaints in regards to "USB Printing" polling serial devices that are not printers, and causing delays in Cura's ability to load model files.  The delays have been described as 5 to 10 minutes.

It isn't at all apparent to the user that the delay is caused by the "USB Printing" as it polls devices that are configured as serial devices, but are not printers, and so the device(s) have no idea what all the "M105"'s mean are that are being sent to them.
There may also be an issue with a device as Cura attempts to alter the port configuration by changing baudrate and possibly rendering the device useless and with the user unaware of what happened and why it happened.

### Describe the solution you'd like

I propose that:
- Cura installs with USB Printing "disabled" by default
- Add a control under "Extensions" that would:
--- Show a checkbox to directly enable or disable the plugin and show the "restart required" dialog if changed.
 Alternatively:
--- Open the MarketPlace and go directly to "USB Printing" where the user could make the selection as they do now.

### Describe alternatives you've considered

N/A

### Affected users and/or printers

Anyone who has additional USB devices routinely plugged into the computer that are not printers, but are serial devices.

### Additional information & file uploads

@fieldOfView and @wawanbreton Any thoughts on this?

edneville, 2024-09-22:
Hello, I've also bumped into this issue. Models load instantly again once this is disabled.

Another alternative could be to offer USB functions once a model is sliced perhaps? This could also (optionally) be enabled after a slice, it would fit into a natural workflow I feel.

Also, wow this is likely the best software I've used in the last few years. Keep up the great work!

Thanks,
Ed

wawanbreton, 2024-09-23:
> Also, wow this is likely the best software I've used in the last few years. Keep up the great work!

Thanks, we do appreciate :)

So we had an other discussion about this recurring topic, and the solution proposed by @Slashee-the-Cow on https://github.com/Ultimaker/Cura/issues/19655 seems to be the best option. We would go with not disabling the whole plugin, which would be a mess, but having an internal plugin setting, disabled by default, and that you would enable e.g. while adding the new printer, or maybe later as suggested by @edneville. This way, the plugin would be completely dormant until it has been manually enabled, not bothering people that don't need it.

For internal reference: CURA-12150

Slashee-the-Cow, 2024-09-23:
> So we had an other discussion about this recurring topic, and the solution proposed by @Slashee-the-Cow on https://github.com/Ultimaker/Cura/issues/19655 seems to be the best option.

Aww, you're making me blush 😊 

> Another alternative could be to offer USB functions once a model is sliced perhaps? This could also (optionally) be enabled after a slice, it would fit into a natural workflow I feel.

@edneville The thing about that is finding a way which *doesn't* stick a "USB" button in the face of everyone who never intends to use USB printing and (since it's internally disabled now) has no reason to disable the plugin.

Maybe if you go to the Monitor tab and the active printer doesn't have USB printing enabled it shows a message:  
*\<Printer name here> does not have USB printing enabled. __Click here__ to enable it.*  
Or my preferred but slightly less user friendly way:   
*\<Printer name here> does not have USB printing enabled. You can enable it in __machine settings__.*  
Yes they have to find the right check box in the machine settings, but now they know where to go to disable it if they want to.

I have no idea about the workflow of an average user, let alone one who uses USB printing, but if you can't find the "Print over USB" button or whatever it says on the Preview tab, is going to the Monitor tab to make sure Cura can see your printer an obvious thing to do?

If the answer to that is "yes" then it could also be the solution for "what to do for existing users who upgrade" problem (if you haven't already figured that out - have USB printing disabled by default but have this obvious way of turning it on - although I don't actually dig through Cura's config files enough to know if it's something you already keep track of).

Unless you know which printers to enable it for automatically then a sticky on the forum saying either "Go into the machine settings for any printers you have set up and disable 'USB printing'" or "Just go to the Monitor tab and the option to enable it" might deter about a third of the people who'd post asking about it 😕

wawanbreton, 2024-09-23:
Thanks for sharing your feedback. I dont't have a strong opinion about the best place to put this button right now, but I'll let our UX designer to make some proposals.

> If the answer to that is "yes" then it could also be the solution for "what to do for existing users who upgrade" problem

That is indeed an issue. We currently don't track the use of the plugin inside Cura, so during the upgrade we will probably have to force-disable it for everyone. I guess the best option in this case is to add a warning in the "what's new" pages so that people don't think it has stopped working. But obviously, not everyone will read it... I'll add this remark as a note to the ticket, so that we can discuss this issue with the team.

GregValiant, 2025-01-19:
Another issue in #20143 .

closed 2025-12-09 by GregValiant