Ultimaker/Cura #19383: Dual infill patterns

spiritparticle, 2024-07-14:
### Is your feature request related to a problem?

I'm frustrated when I want to use an infill type to make the object I am printing stronger internally (like triangles or gyroid) But there are times where that infill completely misses islands that would be printed in the air with no support. On the other hand an infill like "Lightning" accounts for the islands and overhangs but supplies no internal support to make the overall object stronger.

### Describe the solution you'd like

I would like a solution where you can choose TWO infill patterns. For example you could have a gyroid infill to make the print stronger AND a lightning infill also that will make infill supports for islands and overhangs.

### Describe alternatives you've considered

The only alternative I have thought of is to individually add objects and change those objects per object to print as infill to support islands and overhangs. This is meticulous and very tedious.

### Affected users and/or printers

This would benefit ALL users with ALL machines. Having a totally new way to have the best of both worlds. An internal infill that greatly increases the structural integrity of an object that lightning infill does not really do. But then also with lightning infill as a second infill type it can catch islands and overhangs that other infill patterns will not catch because the pattern does not account for it.

spiritparticle, 2024-07-14:
No see this is not at all what I am describing. I want both infill patterns COMBINED. Not the ability to chose one or the other based on which support blocker settings are used. You misinterpreted what I have suggested. For what I am asking your screenshot should show BOTH infill types in the object not one and another bound by object perimeters. The fact that you got what I was suggesting wrong then closed it is also not cool since I had no chance to respond.

Your solution is wrong and not at all what I am describing.

Frederic98, 2024-07-17:
I'd like to suggest to play around with the gradual infill setting, I think this would achieve what you want (but in a bit different way).
- Infill Density = 80%
- Gradual Infill Steps = 2

With these settings, Cura will slice the majority of your object with 20% infill, but increase it to 80% in two steps at areas where it needs to support top layers. This will make sure that "islands" are also always supported by infill.

Shuttle256, 2024-07-22:
@HellAholic 
> > I would like a solution where you can choose TWO infill patterns. For example you could have a gyroid infill to make the print stronger AND a lightning infill also that will make infill supports for islands and overhangs.
> 
> Then please add pictures or some sort of illustrations or use more words to describe your feature accurately. Your request was to have a solution that allows for TWO infill patterns. With the mesh modifier, you can add as many different infill types to as many parts of the model that you desire. You can also use the Tiled infill plugin and create your own custom infill pattern with the shape you desire.

Here is an example of what I believe spiritparticle was talking about; a fused variant:
[image]

spiritparticle, 2024-07-23:
@Shuttle256 has it correct. This is pretty much what I am asking for. It would allow for one type of infill to provide structural support to the overall object. Then choosing another infill like Lightning for example to target any mid air islands that would result from the first infill being too large or missing the island completely, The downside of Lightning however is it results in most of the object being hollow. Being able to leverage both types at the same time would be a tactic to gain structural and island support in a way that would save time and a LOT of filament.

In my first image I want a large infill pattern to give some internal support to my object But notice it does not provide support to the structures. In the second image it shows how lightning infill adresses this but does not do much in the sense of internal support since lightning infill makes most of the object hollow and is very efficient at providing support for overhangs. Now imagine a third image that COMBINES both of these infills as a way to get the best of both worlds.
[image]

My solution would be a way to benefit from an infill like gyroid or triangles to add some internal support to the object where you could add a large infill pattern to save time and filament while still having some kind of internal structure. And then an additional infill such as lightning that will generate infill to the parts only needed by overhangs and islands which ultimately saves more time and filament.

The solution being that this method would contain the best of both worlds. If I only used Lightning infill JUST to address the overhangs and islands then the overall print might be weak due to it being mostly hollow. Using an infill like triangles or gyroid at a small infill density like even 5% gives very good internal structural support but due to the shape and size of these patterns almost always fail to provide support for overhangs and islands. This approach would save a lot of time and filament since it would save you having to print a more dense infill pattern to get support to overhangs and also internal support.