Ultimaker/Cura #19386: Manjaro, CURA won't find OpenGL support, can't connect to d-bus

joe8mofo, 2024-07-15:
### Cura Version

5.7.2

### Operating System

Manjaro Latest

### Printer

Ender 3 Pro

### Reproduction steps

Attempted to run the appimage version 5.7.0. Attempted to run the app image 5.7.2

### Actual results

From commandline: 
qt.qpa.wayland: EGL not available
qt.accessibility.atspi: Error in contacting registry: "org.freedesktop.DBus.Error.Disconnected" "Not connected to D-Bus server"

UI window shows: Could not probe OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers.

 inxi -Fza|grep OpenGL                                                                  1 ✘  35s  
  API: OpenGL v: 4.6 compat-v: 4.5 vendor: intel mesa v: 24.1.1-manjaro1.1

### Expected results

The app should launch.

### Add your .zip and screenshots here ⬇️

[image]
[image]

IlikePepsi, 2024-07-18:
Same issue here.

animal83, 2024-07-24:
Same issue here.

fliegentoeter, 2024-08-02:
Same issue with Cura 5.8 on manjaro...

Entrix22, 2024-08-22:
This has been an issue since since 5.0 seems like they are not willing to fix it.

fliegentoeter, 2024-08-28:
Does anyone know a appimage-version that works?
Tried a few but had no luck...
Or some fork of cura where you can import all printers and settings?

Entrix22, 2024-08-28:
> Does anyone know a appimage-version that works? Tried a few but had no luck... Or some fork of cura where you can import all printers and settings?w
You can install it as a flatpak and it works like that just fine.

fliegentoeter, 2024-09-01:
> > You can install it as a flatpak and it works like that just fine.

And how do I get all my settings for printers and material in? Yes, it is working but without any of my preferences...

fliegentoeter, 2024-09-08:
I tried a few versions and 5.3 is working for me. Just download UltiMaker-Cura-5.3.0-linux-modern.AppImage from: 
github.com/Ultimaker/Cura/releases/tag/5.3.0

All settings and printers are there. But I have no Idea how to backup that stuff and import it in the flatpak-version...
The Cura-Backup-option does not let me make an account and I could not find any setting to export all my settings, printer and material.

davelima, 2024-09-08:
Same happening here with 5.8.1, Fedora 40. The [solution described](https://github.com/Ultimaker/Cura/issues/10815#issuecomment-1783927039) on #10815 is what solved the issue for me.