Ultimaker/Cura #19388: Smarter printing above sparse infill

Hello1024, 2024-07-16:
### Is your feature request related to a problem?

When printing above sparse infill (ie. 5%), there are quite a lot of shapes that fail to print well.

Specifically, the 'skin edge support layers' feature often draws shapes into the infill which are fully unsupported (A), because they're lines that by unlucky chance miss all the infill, or draws paths where part of the path clearly is unsupported (B).

[image]

The end result is that frequently these bits of unsupported printing stick to the print head and make blobs of plastic elsewhere, or even catch on bits of the extruder/fan and knock the whole print off the bed.

### Describe the solution you'd like

These paths need to be made printable in some way.     Consider a line to be printable if both endpoints of a line are within `line_width` of a line on the layer below. 

If a line is unprintable, it should be made to be printable by adding extra support lines on the layer below.  These could form a little like lightning infill - starting with a few basic straight lines a few layers down, to make the area that needs to be supported fully supported where it needs to be.

### Describe alternatives you've considered

n/a

### Affected users and/or printers

all users who print sparsely

GregValiant, 2024-07-16:
I've had this problem myself so these are just some thoughts.  Without increasing the support density (dramatically in some cases) I don't know that this can be fixed.
[image]
The features in the black marks are going to just fall through the grid.  Those might be able to be handled with some sort of fix.
The features in the red shape will fail because you can't print a curve over air, and you can't change directions of straight extrusions over air.  That's a slightly different problem.
I usually address this problem by adding another "floor" (so the plastic has more chance to build up) or going with "Gradual Infill Steps".  Even then some of that is going to be over air.
Without the full structure building up from the bottom, something is going to print over air.  Even the "Stepped Infill" has to print over air each time the density increases.

smartavionics, 2024-07-16:
How about a setting that would force an infill layer to be printed as, say, 50% density zigzag pattern when the first layer of infill support is detected? Alternatively, you could have a setting that specifies that a particular layer (or layers) were infilled using a 50% zigzag?

GregValiant, 2024-07-16:
This is with @smartavionics idea but using an "Infill Mesh" configured for zig-zag at 50%.  This looks like it would work for the problem I was having.
[image]

smartavionics, 2024-07-17:
Checkout layers 13 and 36...

https://github.com/user-attachments/assets/ef6f55f0-6e39-44d1-9759-55cad93c8aeb

GregValiant, 2024-07-17:
That'll work.  You just released 4.20.22 though.  Any timeline for the next version?

smartavionics, 2024-07-18:
First week of August? If you would like it sooner, I can do that.

GregValiant, 2024-07-18:
Your schedule is fine.  At one point I was printing a lot of models that had lettering on top of a base and the problem was really apparent as there were a lot of curves and line segments that tried to change direction over air.  For that most part the problem could be abated by increasing the Skin Expand Distance but that isn't always a good option.

Hello1024, 2024-07-18:
https://github.com/user-attachments/assets/fbbcf49c-5165-43d4-929c-9bf666e135d2

Came up with a different solution here that after a few test prints I thinks better...

The idea is to add specific lines between infill to make sure outlines of the skin above will be fully supported.   With this, it no longer seems to be necessary to have `Skin Expand Distance` or `Skin Edge Support` - I suspect both of those features were really trying to solve this problem.

It works by making a 'low poly' version of the skin to figure out the minimum points that need supporting (recognising that a curve cannot be printed over thin air), and then adding lines on the layer below to support those points.

It handles thin surfaces caused by curved shapes better, and uses less filament (important for low density parts).

Thoughts?

In development here:  https://github.com/Hello1024/CuraEngine/tree/Hello1024-better-infill-support

smartavionics, 2024-07-18:
I can't view your video as my browser says the file is corrupt. However, your plan sounds pretty good.

Hello1024, 2024-07-18:
Youtube link to video:
  https://youtu.be/eJfT7keC-KE

Screenshot of a layer about to support a very oddly shaped surface:
[image]

The oddly shaped surface:
[image]

Todo things before it's a PR ready to go:

  * Some kind of test
  * Check it works with all kinds of infill  (probably doesn't right now - infills like lightning make huge bounding boxes)
  * Check it works with gradual infill
  * Check it works with multiple parts and infill modifiers
  * The design involves complex shapes and iterating points - go find big/complex files to test and be sure there are no performance regressions
  * Decide if it is a suitable replacement for the `Skin Expand Distance` or `Skin Edge Support` options, or if it should be added as another setting, and if so, how should the settings interact if all are enabled.
  * Decide if its worth implementing some possible extensions:
    * Support more than just skins - there are places where walls also need support, for example when skins are disabled.
    * Support the *middles* of some skins - for example, support the end of every line within the skin, rather than just the skin boundary.

Hello1024, 2024-07-29:
Ultimaker/CuraEngine#2122 fixes this, and is ready for review.

[image]

On the left is the latest git release.   On the right is Ultimaker/CuraEngine#2122.    The print has been stopped at exactly the same point after just 1 layer of skin.

Notice how there are holes in the skin on the left, whereas the 0.15mm layer is nearly perfectly hole free on the right.   All other settings are the same.

For this test model, with all 3 options leading to similar skin surface quality:

| Time       | Volume (mm^3) | Description                                                      |
|------------|---------------|------------------------------------------------------------------|
| 28m 54s    | 3142          | With this feature (Extra Infill Lines to Support skins = Walls and Lines mode) |
| 30m 43s    | 3211          | With 4 layers Skin edge support                                  |
| 33m 38s    | 3395          | With 2 extra skin layers                                         |

So it seems this leads to faster prints, less filament used, and same surface finish than other approaches.