Ultimaker/Cura #19390: spawn tree on bridge?

Scull215, 2024-07-17:
### Is your feature request related to a problem?

getting supports where needed while trying to avoid spawning them on model

### Describe the solution you'd like

on tree supports could we have 2 seperated trees get a bridge between them to get a new tree generated on the bridge probably over a model to avoid the tree on model and still get the support needed while maby saving material in some cases 

### Describe alternatives you've considered

high angled tree supports

### Affected users and/or printers

it should be everyone

### Additional information & file uploads

I thought of this while slicing a phone case and the screen side lip that would hold the case/phone together I would be able to support about half without anything on the case other than the overhangs
I'm assuming this would require tuning on the users part like bridge thickness and bridging in general to hold the tree

GregValiant, 2024-07-17:
I think this could use some sort of sketch to explain exactly what you are asking for.

Scull215, 2024-07-20:
even my stick figures are bad so I'm having the artistic wife sketch something up although she's not into 3d printing

ThomasRahm, 2024-07-23:
> I think this could use some sort of sketch to explain exactly what you are asking for.
The idea has come up in the past.
They want something like this:
[image]

Which seems like a good idea. In reality it would most likely result in something like which is much less stable.
[image]

Not saying it wouldn't work, it should in some cases.
Basically one would need to check which areas inside an avoidance can be reached by printing a bridge between two branches outside the avoidance, and continue avoidance generation from there.
I suspect that there would need to be another class of avoidances (performance impact!), to prevent this bridge supporting from occurring to often.
When such bridge support is placed, basically there would be two new tips added that are connected with a bridge, on which another branch rests.
As the tree support only generates support areas, and not lines, this direction between the tips would need to be somehow saved.
It would only help for lowish height holes, that need support in the middle, as such I suspect the use-case to be quite limited. 
If I ever encounter an idea to prevent additional avoidances and therefore limit the performance impact that could be something worth looking into.

Tl;dr: Performance impact and annoying to get the line direction to the line generation is why I didnt look into doing something like this.

GregValiant, 2024-07-23:
Thanks for the thoughts on this.
I attempted to put in a model of the bridge and set it to Print as Support.  The support location set to "Build Plate Only" seemed to be getting in the way.  No support would generate in the "exclusion zone" under the overhang.
[image]

Switching to Everywhere allowed the support to generate but it isn't at all what I had wanted.
[image]