Ultimaker/Cura #19398: Cura 5.8b1 Print tries to start as soon as bed reaches temp

LilBub, 2024-07-18:
### Cura Version

5.8.0 Beta 1

### Operating System

Windows 11

### Printer

Ultimaker 2+

### Reproduction steps

Slice print
Upload to latest Octoprint
Noticed that as soon as temp hit 60 on bed, print start
I also observed that it did not try to do any of the pre print material purge

### Actual results

Print tries to print with a cold hot end once bed is at 60c

### Expected results

Normal print!

### Add your .zip and screenshots here ⬇️

[UM2_Test_cube_100_flow_Proj_cura58b1.zip](https://github.com/user-attachments/files/16290106/UM2_Test_cube_100_flow_Proj_cura58b1.zip)

I don't know if Octoprint is mucking with the code one uploaded or if this Cura.

I was able to repeat the same exact issue with Cura 5.7.2

LilBub, 2024-07-18:
Checking Octoprint, there are no GCODE start commands 

[image]

LilBub, 2024-07-18:
I just tried a print from a few months ago stored on my Octoprint Raspberry Pi that was made with Cura 5.7
That's printing fine. 
So, something is up with the newer version(s)

Asterchades, 2024-07-19:
OctoPrint's start and end scripts are completely independent of Cura's, and are quite often left blank so as not to interfere with the slicer you've used. Only the Cura scripts will actually be embedded in the resulting GCode.

From what I can see, it looks like the automatically generated temperature commands embedded in the GCode (so only influenced by Cura) is not creating an M109 statement to wait for the tool to finish heating. It's setting the tool to heat (M104) - just not waiting for it to finish, which in turn will likely have prevented it from purging due to low temperature failsafes in the firmware. 5.7.2 broke part of this (specifically the ability to override it) so work has been done here, and it would seem that perhaps something was inadvertently broken in the process.

Someone on the team will take a look at this.

GregValiant, 2024-07-19:
@LilBub Your StartUp gcode doesn't contain any Cura Keywords for the bed and hot end temperatures.  That means that the temperature "prepend" will be put in.  There are two settings that also control the content of the Prepend in regards to "waiting".  You must load the "Printer Settings" plugin from the MarketPlace to access them.
[image]

That's from your project and you have "Wait for Build Plate Heatup" checked and so Cura adds the M190 line.  But "Wait for Nozzle Heatup" is NOT checked (that is the new default and a result of the fix for the 5.7.2 problem) and so only the M104 line is added and the "wait" command M109 is not.
When you slice and save the project you get this:
M140 S60
M105
M190 S60
M104 S210
M82 ;absolute extrusion mode
That does not have the M109 line.

With "Wait for Nozzle Heat Up" checked you would get this:
M140 S60
M105
M190 S60
M104 S210
M105
**M109 S210**
M82 ;absolute extrusion mode

So the default went from "Enabled" to "Disabled" and so you get what you got.  It is "expected behavior" given the repair that was made.
I'll remove the bug label and I'm inclined to close this.  Let us know if checking that setting works for you.
You could also alter your startup gcode to include the Cura Keywords, but simply making that setting adjustment should work.  If you have custom setting profiles they may need updating to include the "enabled" for the setting.

LilBub, 2024-07-19:
Hi,

A few points

I don't recall changing that recently since Octoprint needs Marlin.  I am looking through previous versions and I had G-Code flavor set to the Marlin in 5.7.0 and things worked fine.  I also see the identical settings (Marlin) in 5.6.0.

So - don't be so quick to shut down my ticket quite yet.

Also, this check box to Wait for Nozzle doesn't show up for me

[image]

If I drill down into the settings, I see it there but it has a icon in front of it

[image]

GregValiant, 2024-07-19:
I'll stick the bug label back on.
I went through the U2 definition and the U2Plus definition and there are no over-rides for "Wait for nozzle heatup".
Going deeper - In order for it to be visible "Enable Nozzle Temperature Control" must be checked and in order for that one to be visible ""machine_gcode_flavor != \"UltiGCode\"" which is Greek for "The Gcode flavor cannot be "Ultimaker 2".
So if you switch back to Marlin, it will be visible.
BUT
as @HellAholic stated:  _"Switching to Marlin for UM2+ is not recommended and not officially supported."_
I can understand that you want to use Octoprint and it may be possible.  Adjusting your StartUp gcode to include:
M109 S{material_print_temperature_layer_0}
might be something to try.

@wawanbreton Everyone else has kicked in...can you shed any light on this?  It would appear that if the firmware is set to Ultimaker 2 that the hot end prepend temperature is not being applied (because the firmware handles it(?) or maybe because UM2 firmware isn't playing nice with Octoprint(?) or (most likely) I'm just having a senior moment and everything is OK.

LilBub, 2024-07-21:
@HellAholic , my Ultimaker 2+ "Machine Settings" were already it set to Marlin mode.

[image]

What has me confused is I upgraded another PC (desktop) in my workshop to Cura 5.8 B1
Upon updating and installing the Printer Settings Plugin for it (to match the laptop), I checked a few things.

1) Same Marlin settings (Marlin)
2) Now this is strange, when I look at the Printer Settings, on the dekstop install, I see the check box for **Wait for Nozzle Heatup**
[image]

Why I can see it on one install and not another has me stumped

At first, I thought it might be the Profile loaded (Nomal - 0.15mm).  On each install I changed the material to Ultimaker PLA just to match up.  I would have expected the pulldown to offer several options but I am used to only seeing one.  (Save that question for later)

That said, I'm curious why one has the check box, and one does not?

LilBub, 2024-07-21:
> Indeed that is puzzling. Could you maybe make a zip out of your configuration folder on the device that doesn't show the setting and send that over?

Specifically which folder?
[image]

LilBub, 2024-07-21:
25mb barrier, Zip files is 26mb   Anything you do not need?

LilBub, 2024-07-21:
Sure!

[image]

fieldOfView, 2024-07-22:
Just for the record, @HellAholic :

> Switching to Marlin for UM2+ is not recommended and not officially supported.

This is untrue. If you want to use the UM2 family to print via USB (which includes using Octoprint, because that ultimately prints over USB), you MUST switch to the Marlin Gcode flavor. This has been true ever since the UM2 was launched. The "Ultimaker 2" gcode flavor does not have start gcode (because that is added by the printer firmware). When switching an Ultimaker 2 family printer to the "Marlin" gcode flavor, a good starting point for Start- and End gcode snippets is automatically activated (and can be edited in Machine Settings). I worked on this extensively when I was working on Cura 2.x. 

The OctoPrint plugin specifically checks if Ultimaker 2-family printers are using the "Ultimaker 2" flavor, and offers to make the change to "Marlin" for the user, because that is the only way Ultimaker 2-family printers can be used with OctoPrint. See also https://github.com/fieldOfView/Cura-OctoPrintPlugin?tab=readme-ov-file#notes-on-ultigcode-ultimaker-2ultimaker-2

If you intended to say that printing over USB (via OctoPrint or not) is not recommended officially supported, that may be true.

LilBub, 2024-07-22:
@HellAholic , just to provide some more information - for the past 4 days I have had 9 machines running non stop using laptop, slicing a variety of prints that all work with Octoprint without any issue.  With the newer versions of Cura, I can't utilize my Ultimaker 2+ via Octoprint due to this issue.   

So, I don't think it is necessarily a plugin issue.  (But I'm willing to try your request to evaluate if the Printer Settings / Wait for Nozzle is remedied by toggling off some plugins)

LilBub, 2024-07-22:
Hi @HellAholic , when I get home from work today I'll offer a Google Drive link for that.

LilBub, 2024-07-23:
@HellAholic  here you go

https://drive.google.com/file/d/1SynurHW5_l_G6wYqsDnh2pfPDKkqTYxf/view?usp=sharing

LilBub, 2024-07-23:
You should have access to that link (I thought you already had it!)

LilBub, 2024-07-23:
Well on my end it shows Marlin so that's kinda wonky?

LilBub, 2024-07-23:
Now with those settings, do you see the Nozzle heat up option?

LilBub, 2024-07-23:
I have all the versions, but yes, I had 5.7.2 and upped to 5.8 B1

LilBub, 2024-07-24:
Ok I know why you are seeing Ultimaker 2 in the Marlin type.
You had suggested changing that to see if the Wait for Nozzle Heatup would show up in Printer Settings  :-)

LilBub, 2024-08-06:
Just chiming in to update - I DO see the Heat Up Nozzle option now and I am able to print via Octoprint to my Ultimaker 2+