Ultimaker/Cura #19400: Error de slicing en Cura 5.6

GPCsoftware, 2024-07-19:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Hellbot Magna 2 230

### Name abnormal settings

Capas adaptativas

### Describe your model

iiii

### Add your .zip here ⬇️

[1.zip](https://github.com/user-attachments/files/16315504/1.zip)
[image]
[image]
[image]

LilBub, 2024-07-19:
Let me try via Google Translate

Just trying to be helpful posting here; have you tried the newer version of Cura, 5.7.2?  Also, you file contains only screenshots.  A project file would be useful.

Solo intento ser útil publicando aquí; ¿Has probado la versión más nueva de Cura, 5.7.2?  Además, su archivo contiene solo capturas de pantalla.  Un archivo de proyecto sería útil.

GregValiant, 2024-07-19:
Cura 5.6.0 is 4 versions behind.  The Cura 5.8 beta appears to be very good and does not have the slicing issues that 5.5 and 5.6 had.

closed 2024-08-03 by github-actions[bot]