Ultimaker/Cura #19406: can't slice anything

Siclou529, 2024-07-20:
### Cura Version

5.7.0 bata

### Operating System

Big Sur

### Printer

ender 3v2

### Name abnormal settings

nothing slices not even temp towers 

### Describe model location

no

### Describe your model

yes

### Add your .zip here ⬇️

[Fender Stratocaster Mini guitar model - 2748193.zip](https://github.com/user-attachments/files/16322049/Fender.Stratocaster.Mini.guitar.model.-.2748193.zip)

GregValiant, 2024-07-20:
Thanks for the report.
Have you tried a newer version of Cura?  5.8 beta is out now so 5.7beta is 3 versions behind.
We would need a project file and your Cura log file to find out what was going on.

Since 5.7 beta is obsolete I'm going to go ahead and close this.  If you have trouble with a more up to date version then please put in another request (with project file and Cura.log file).

closed 2024-07-20 by GregValiant

Siclou529, 2024-07-20:
Yes I did try the newest version same problem was working fine thinking of resettling my computer and reinstalling cuts. Sent from my iPhone