Ultimaker/Cura #19408: Part corner warping

Zark99, 2024-07-20:
### Is your feature request related to a problem?

When im printing some parts the initial layer, normaly the outer wall, or a circle in the middle of the part, dont adhere to the bed.
The bed is leveled and the z ofset is perfect.

### Describe the solution you'd like

To solve the problem i manualy change the z offset whille the print is running. I change the z offset only on the bottom outter walls to squish that first layer. After that, i change it back to the optimal z offset and the print will never warp.
It would be nice if the slicer could the initial outter wall height so it would adhere better. 

[image]

### Describe alternatives you've considered

Decreasing first layer speed, the problem continued. Increase the z offset (more squish), solved it but the extrusion would turn out ugly, scraped but the extruder being to low. 

### Affected users and/or printers

It will affect users that have warped prints, has it increases adhesion on the outer most layer

Zark99, 2024-07-20:
In the picture we can see the outer wall with perfect adhesion due to lower z offset(-0.25). We can then see the inside of the outer walls, if we stay at that z offset the print will be scraped(Left Side of thr Print) and if we then change to a higher z offset(-0.10) the print will be perfect (Right Side of thr Print).

GregValiant, 2024-07-20:
This sounds like it's really a printer specific problem.  I have the same bed on my Ender and I have never had this happen.
Adjusting the Z offset has the same effect as increasing the Flow Rate.
Having to adjust the Z offset would seem to indicate that the bed is wavy, or something.  If it was exactly flat there should not be a problem.
Although you can adjust the "Intial Layer Flow" to 105% or 107% in Cura, there is no setting for "Outer Wall Initial Layer Flow".
Adjusting the Z offset on the fly during the first layer is probably doable in the gcode (M851?), but I have to believe there are better ways to do it.

LilBub, 2024-07-21:
Is the first layer set to be a bit thicker?  That could help if your bed is a tad wavy, as Greg points out.
I've lost track of all 1,257 versions of the Ender - is that one using a BL Touch and creating a mesh? (G29)
Are you Babystepping the first layer?

Zark99, 2024-07-21:
It has manual mesh bed leveling (4x4). 
First layer is set to 0,28mm.
I babystep until get the first layer perfect, then if i try to print again looks like the moment it starts printing the first line, its to far and wont stick for a few mm (sometimes not just a few). It looks like the filament doenst have time to drop to the bed before it stars moving. But usualy only happens on the first line (the countour of the part), for the rest it adheres just fine.
https://github.com/user-attachments/assets/5ff7e66b-079c-4baa-a971-368eeec64402

LilBub, 2024-07-21:
Hi,
Thanks for the video.  Like most Creality machines the middle of the bed is lower than the outside.  4-5 seconds into your print and I can see how high that nozzle is above the bed.  I bet that bed clip in the middle helps with this.    A BL Touch would build a mesh and help with this a good amount.

But from what I see as a guy with 23 machines that print a silly amount of build projects....this isn't a Cura issue.

GregValiant, 2024-07-21:
Cura (any slicer) never knows the internal printer settings and so Cura never knows what the "real" initial Layer Height" might be.  It can only do the calculations based on the numbers in the Cura settings.  So the E values for the Initial Layer are actually theoretical rather than real.
"Initial Layer Height x Line Width x Length of Extrusion" = "Volume of Extrusion".
If you change the Initial Layer Height by bouncing the Z up and down, then you are changing how much plastic is required for the extrusion.  The plastic never"drops to the bed" but rather it is "squished into the bed" to make a rectangular shape that is Layer Height tall x Line Width wide.
If you change the Z of the initial layer like from 0.28 to 0.20 then you are effectively over-extruding that area by about 30%.  That'll stick I betcha.
But now the second layer comes along and it was also calculated at .28 layer height, but when it gets to the dip caused by the Initial Layer Z adjustment, now it's under-extruding and won't stick because instead of a .28 layer height in the dip it is a .36 layer height.  So that area under-extrudes by about 30% and has trouble adhering to the first layer.

I think what you have here is a printer problem and it needs to be solved at the printer.

Zark99, 2024-09-01:
Bought a 3D Touch from Aliexpress, and dishwashed the bed with some hot wather. Prints like a charm. I think the problem was the invisible uncleness on the bed rather then the leveling.

closed 2024-09-01 by Zark99

GregValiant, 2024-09-02:
Nice that it's fixed.