Ultimaker/Cura #19411: Slicer starts slicing then just never finishes.

MrRickie, 2024-07-21:
### Cura Version

5.8.0

### Operating System

windows 11

### Printer

Lotmaxx Shark

### Reproduction steps

Loaded stl from drive then started with checking printer and slicer settings all look good then start slicing. 

### Actual results

Slicer starts slicing then just never finishes. Waited over night and it never went past half way. It just starting doing this was working fine before. Have not changed anything sence last GOOD slice.

### Expected results

Slicer should slice the stl file then create a gcode file to save to disk.

### Add your .zip and screenshots here ⬇️

[image]
[image]

gazhay, 2024-08-06:
Can confirm this is the case for the last few Linux appimage release too. 

This file never slices
[Francis.zip](https://github.com/user-attachments/files/16509606/Francis.zip)

gazhay, 2024-08-16:
> @gazhay I recommend you look into the user changed settings and configurations you have.
> The file you included is a STL rather than a 3mf so I have no way of reproducing the issue you're reporting.
> To break your absolute statement of "This file never slices", here is a sliced version of the STL you included using the 5.8 appImage. I picked a random printer.
> [image]
>  

Well previous versions handle the file with those settings, its a thing in recent versions (as in settings & file don't change but behaviour does)

My workflow has always been (6 or 7 years+) openscad->stl->cura->scad->print so i've no idea what 3mf I would need to produce?! Seems like a red herring in this case.

closed 2024-08-27 by github-actions[bot]