Ultimaker/Cura #19413: Fan G-code starting Before, gives randoms values to the gcode after slicing it

mystermonster94, 2024-07-22:
### Cura Version

5.7.2

### Operating System

windows 11

### Printer

artillery sidewinder x1

### Reproduction steps

1. I put the fan to start in the after 5mm, like before the 30 layer... but it starts at the first row, it just do the first layer, and then the other all layers give randoms values to up... it dosnt start at the 30th layer, it starts at the second... i check the gcode im gona put the gcode down below 

### Actual results

;LAYER:1
M106 S1.8
;TYPE:WALL-INNER
;MESH:perno de 8mms.STL
G1 F1500 E28.03692
G1 F2850 X150.326 Y155.965 E28.04953
G1 X151.069 Y155.884 E28.07687
G1 X151.799 Y155.735 E28.10413
G1 X152.155 Y155.742 E28.11716
G1 X152.365 Y155.561 E28.1273
G1 X153.126 Y155.219 E28.15782
G1 X153.802 Y154.832 E28.18632
G1 X154.121 Y154.738 E28.19849
G1 X154.31 Y154.465 E28.21064
G1 X154.878 Y153.944 E28.23884
G1 X155.429 Y153.344 E28.26864
G1 X155.663 Y153.203 E28.27864
G1 X155.725 Y152.933 E28.28877
 first layer and it starts with some random value... 
second to the 5 layer..

;LAYER:10
M106 S127.5
;TYPE:WALL-INNER
;MESH:perno de 8mms.STL
G1 F1500 E187.42233
G1 F2966.6 X150.326 Y155.965 E187.43494
G1 X151.069 Y155.884 E187.46228
G1 X151.799 Y155.735 E187.48954
G1 X152.155 Y155.742 E187.50257
G1 X152.365 Y155.561 E187.51271
G1 X153.126 Y155.219 E187.54323
so u can see, it start in the 1 layer, when it actually should start at the 30th layer...

### Expected results

i change by myselft the g-code one by one..

;TIME_ELAPSED:32.650427
;LAYER:1
M106 S0
;TYPE:WALL-INNER
;MESH:perno de 8mms.STL
G1 F1500 E28.03692
G1 F2850 X150.326 Y155.965 E28.04953
G1 X151.069 Y155.884 E28.07687
G1 X151.799 Y155.735 E28.10413
G1 X152.155 Y155.742 E28.11716
G1 X152.365 Y155.561 E28.1273
G1 X153.126 Y155.219 E28.15782
G1 X153.802 Y154.832 E28.18632
G1 X154.121 Y154.738 E28.19849
G1 X154.31 Y154.465 E28.21064
G1 X154.878 Y153.944 E28.23884
G1 X155.429 Y153.344 E28.26864

G0 F10500 X150 Y156.077
;TIME_ELAPSED:52.216463
;LAYER:2
M106 S0
;TYPE:WALL-INNER
;MESH:perno de 8mms.STL
G1 F1500 E45.55965
G1 F2964.8 X150.326 Y155.965 E45.57226
G1 X151.069 Y155.884 E45.5996
G1 X151.799 Y155.735 E45.62686
G1 X152.155 Y155.742 E45.63989
G1 X152.365 Y155.561 E45.65003
G1 X153.126 Y155.219 E45.68056
G1 X153.802 Y154.832 E45.70905
G1 X154.121 Y154.738 E45.72122
G1 X154.31 Y154.465 E45.73337

G0 F15000 X150 Y156.077
;TIME_ELAPSED:71.739102
;LAYER:3
M106 S0
;TYPE:WALL-INNER
;MESH:perno de 8mms.STL
G1 F1500 E63.08312
G1 F2964.5 X150.326 Y155.965 E63.09573
G1 X151.069 Y155.884 E63.12308
G1 X151.799 Y155.735 E63.15034
G1 X152.155 Y155.742 E63.16336
G1 X152.365 Y155.561 E63.17351
G1 X153.126 Y155.219 E63.20403
G1 X153.802 Y154.832 E63.23253
G1 X154.121 Y154.738 E63.2447
G1 X154.31 Y154.465 E63.25685
G1 X154.878 Y153.944 E63.28504
G1 X155.429 Y153.344 E63.31485
G1 X155.663 Y153.203 E63.32484

### Add your .zip and screenshots here ⬇️

[image]

mystermonster94, 2024-07-22:
> @mystermonster94 , we fixed a bug related to the fan speed in 5.8-Beta, could you check if you still see the issue in that version?

i download and test the 5.8-beta version and still failing it starts in the 1 layer the fan... when it should start at the 30th layer.... omggg this is getting my crazy, can someone solve this pls

[image]

G0 F300 X147.092 Y143.5 Z0.72
G0 F10500 X137.74 Y135.011
;TIME_ELAPSED:77.395449
;LAYER:2
M106 S3.5
;TYPE:SKIRT
G1 F1500 E70.94158

mystermonster94, 2024-07-23:
i guess i have to say goodbye to ultimaker cura, no development in this program. no solution

gh13, 2024-07-23:
Cura ramps the fan speed from the initial speed specified from the 1st layer to the regular fan speed at the specified layer/height.
If you want the fan to start at 5mm you can use the Advanced Cooling Fan Control Post-Processor to set a custom fan profile for your requirements

[image]

mystermonster94, 2024-07-23:
> Cura ramps the fan speed from the initial speed specified from the 1st layer to the regular fan speed at the specified layer/height. If you want the fan to start at 5mm you can use the Advanced [...]

It was working 1 week ago and it crashed.. is stupid to ramp a simple 5mm+ height to starting fan, this is a development problem.  I dont Need diferent layers to be fan just the normal height to start the fan.. this nonsense bug, the idea of the cura is make working simple work flow not adding more problem for the users.

gh13, 2024-07-23:
Something is strange with your settings as it should not be possible to have a regular fan speed at height 0.0 mm & regular fan speed at layer 30 because when you change one of the settings the other should automatically change to the corresponding value based on your layer height. Maybe your profile has become corrupt due to the crash & you may have to try to recreate the profile from a default profile

mystermonster94, 2024-07-24:
> Something is strange with your settings as it should not be possible to have a regular fan speed at height 0.0 mm & regular fan speed at layer 30 because when you change one of the settings the [...]

i just delet and unistall cura slicer, but the profiles keep appearing when i install again, how do i delete even the profiles to re install again the cura like new, to see if it working... OMG i dont know how this bug appear but its getting my crazy dude.

gh13, 2024-07-24:
you can either delete the profile by activating another profile & delete the profile in mange profiles.
Or before you do that; activate one of the default profiles, modify it to your requirements & save as a new profile then if it works delete your old profile

mystermonster94, 2024-07-24:
it dosnt work anything....

gh13, 2024-07-24:
What doesn't work anything? Deleting the profile, creating a new one or have you created a new profile & it doesn't work?

mystermonster94, 2024-07-24:
> What doesn't work anything? Deleting the profile, creating a new one or have you created a new profile & it doesn't work?

yeah i just delete all the profiles, and use the standar profiles, and still dosnt working... it keeps starting the fk fan at the first layers, not from the layers when it should.... i unistall all version of cura, to install the 5.8 beta last one.. and still dostn working... omg i cant solve this, this get me crazy. it was working 1 week ago. f cura

gh13, 2024-07-24:
I'm just a user like yourself & out ideas. Maybe one of the devs has more of an insight into finding the cause
Hope you find your solution. Sorry I'm unable to help any further

gh13, 2024-07-24:
Final idea.... You could go here... go up a folder level, Rename or Delete the 5.8 folder. Close & Open Cura, which should recreate the 5.8 folder with a default configuration & give you a clean slate to work from. You will have to set up your printers, preferences & profiles again but I can't think of anything else

[image]

GregValiant, 2024-07-25:
I'm going to go ahead and close this as it appears to be a duplicate of your other report #19429.  I hope you understand.

closed 2024-07-25 by GregValiant