Ultimaker/Cura #19414: falla de lectura de archivo gcode

AgustinAban, 2024-07-22:
### Is your feature request related to a problem?

problema de lectura en el archivo gcore a la hora de conectar la tarjeta de memoria en la impresora no me figura el archivo gcore en la impresora. tampoco puedo imprimir conectando por cable la compu a la impresora, se manda la señal de imprimir pero me imprime fuera de la cama de impresion, no se cual es la configuracion que tendre que hacer ya que no me sucede con otras computadoras, probe instalando versiones anteriores pero el resultado es el mismo no figura el archivo o me imprime fuera de la cama de impresion, desde ya muchas gracias por su ayuda. saludos.

### Describe the solution you'd like

creo que configurando algunos parametros ya sea del programa o de la pc se solucionaria el problema.

### Describe alternatives you've considered

probe con versiones anteriores y las mas nuevas, tambien conectandolo mediante usb a la impresora

### Affected users and/or printers

afectaria a usuarios con el mismo problema

GregValiant, 2024-07-23:
If both the SD card fails and the USB printing fails then the problem must be with the mainboard of the printer.  One way or the other ought to work.  It would be very odd for both the SD card slot and the USB port to fail at the same time, but if something in the chipset died, then it would affect both ways of printing.
If you move the nozzle using the LCD control does the print head move as it should?
Try reformatting the SD card and then copy a gcode file to it and try printing it.
Printing "Outside the Print Bed" can have a couple of causes.  One would be if "Origin at Center" is checked in the Cura machine settings, but the printer origin is at the left from corner.  Another reason would be if the mainboard settings have become corrupt and the Home Offset setting is way off.
What printer is it?

GregValiant, 2025-04-07:
I don't see a message there.  Can this be closed?

closed 2025-04-21 by github-actions[bot]