Ultimaker/Cura #19423: `Generated with Cura_SteamEngine` persist on top of custom start gcode

PingRen32, 2024-07-24:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

Ender 3

### Reproduction steps

I got a starter code 

```
; Ender 3 Custom Start G-code
M140 S{material_bed_temperature_layer_0} ;Start heating bed
M190 S{material_bed_temperature_layer_0} ;Wait for bed to reach temp before proceeding
M900 K0.11; PLA PRO Linear Advance
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; Bed leveling
M104 S{material_print_temperature_layer_0} ;Start heating extruder
M109 S{material_print_temperature_layer_0} ;Wait for extruder to reach temp before 
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
```

which should skip the `Generated with Cura_SteamEngine` (which did back from 5.3) so something must have been broken.

### Actual results

```
;FLAVOR:Marlin
;TIME:2018
;Filament used: 1.39217m
;Layer height: 0.2
;MINX:96.2
;MINY:96.2
;MINZ:0.12
;MAXX:123.8
;MAXY:123.8
;MAXZ:19.92
;TARGET_MACHINE.NAME:Creality Ender-3 Pro
;Generated with Cura_SteamEngine 5.7.2
M104 S195
M105
M109 S195
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
M140 S65.0 ;Start heating bed
M190 S65.0 ;Wait for bed to reach temp before proceeding
M900 K0.11; PLA PRO Linear Advance
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; Bed leveling
M104 S195.0 ;Start heating extruder
M109 S195.0 ;Wait for extruder to reach temp before 
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
```
It keep adding `Generated with Cura_SteamEngine` back.

### Expected results

No extra header added on top.

### Add your .zip and screenshots here ⬇️

[CE3PRO_XYZ_20mm_Calibration_Cube.zip](https://github.com/user-attachments/files/16358822/CE3PRO_XYZ_20mm_Calibration_Cube.zip)

Just a basic test cube.

GregValiant, 2024-07-24:
Thanks for the report.
That line has been there as at least as far back as Cura 2.3.  It's a comment so has no effect on the printer unless the printer is specifically looking for it.  Post processors might use it if they are version specific.  It is always the first line of the StartUp section of the gcode, followed by the Machine StartUp Gcode (if there is any) and then an extruder reset and retraction (when enabled) and then the "LAYER_COUNT:" line finishes that Gcode section.  All forks of Cura have that line:

;Generated with Sovol Slicer 1.7.2

This is expected behavior so I'll remove the bug label and close this.

closed 2024-07-24 by GregValiant

PingRen32, 2024-07-25:
This is a issue still need addressing @GregValiant

GregValiant, 2024-07-25:
OK, I understand.
5.7.2 had a bug that added the "prepend" temperature even if {material_print_temperature} or {material_print_temperature_layer_0} are in the StartUp G-Code.  That has been fixed in 5.8beta.

To fix the problem in 5.7.2:
Download [CuraPrependBugFix](https://github.com/user-attachments/files/15806103/CuraPrependBugFix.zip), unzip the file, and place the ".py" file into your "Configuration Folder" ("Help | Show Configuration Folder") and the "scripts" sub-folder.  It will be available under "Extensions / Post Processing / Modify Gcode" and then "Add Script".  When "Cura Prepend Temperature Bug Fix" is enabled it will remove those temperature lines.

I'll put the bug label back on this and mark it as a duplicate.  I'll leave it closed since the problem has been fixed in 5.8beta.