Ultimaker/Cura #19425: when i will load a file. Cura crashes before a dialogbox appears,

artbody, 2024-07-24:
### Cura Version

5.7.2

### Operating System

Gentoo Linux

### Printer

DIY smothieboard based 

### Reproduction steps

1 Starting Cura

2 File -> Open File = CRASH

### Actual results

Hello

in a xterm i start cura
it starts as expected, but when i will load a file it crashes after a few seconds

UltiMaker-Cura-5.7.2-linux-X64.AppImage
Fontconfig warning: "/etc/fonts/conf.avail/05-reset-dirs-sample.conf", line 6: unknown element "reset-dirs"
qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///tmp/.mount_UltiMawBjojQ/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMawBjojQ/share/cura/resources/qml/Actions.qml"
Speicherzugriffsfehler (Speicherabzug geschrieben)

uname -a
Linux artbody 6.6.13-gentoo-x86_64 #1 SMP PREEMPT_DYNAMIC Fri Mar 1 10:49:26 CET 2024 x86_64 AMD Ryzen 9 5950X 16-Core Processor AuthenticAMD GNU/Linux

best regards
Achim

### Expected results

Opening a filedialogbox

### Add your .zip and screenshots here ⬇️

[cura.log.zip](https://github.com/user-attachments/files/16363044/cura.log.zip)

vo-omega-001, 2024-07-24:
Hi,
I reproduice the same issue in docker container based on mini debian 12,
from  Cura github sources: SHA-1: 935109680357710057f8bc1b6cf2dec3e9d317da

<b>I 've RESOLVED this issue with the following env var : QT_QPA_PLATFORMTHEME="gtk3" </b>

by default, QT_QPA_PLATFORMTHEME="xdgdesktopportal" and a crash occurs when PyQt6 FileDialog popup is loaded.

Cura/venv/lib/python3.10/site-packages/PyQt6/Qt6/lib/libQt6LabsFolderListModel.so.6  libQt6LabsFolderListModel.so.6 is never loaded 
but  Cura/venv/lib/python3.10/site-packages/PyQt6/Qt6/qml/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml requires it (cf: import Qt.labs.folderlistmodel)

NB: FileDialog.qml is called by Cura/resources/qml/Cura.qml

vo-omega-001, 2024-07-26:
When I launch Cura AppImage on Ubuntu 24.04, it works fine with the following steps:

1. sudo apt install -y qt6-xdgdesktopportal-platformtheme
2. export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6
3. sudo ./UltiMaker-Cura-5.7.2-linux-X64.AppImage -platformtheme gtk3