Ultimaker/Cura #19427: No slicing after update from 5.30 to 5.7.2

reuschi, 2024-07-24:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

Anycubic Vyper

### Name abnormal settings

No slicing at all - even not with default profile(s)

### Describe model location

No slicing at all

### Describe your model

I sliced it successfully in Cura Version 5.3.0. Then updated Cura and error is happening all the time.

### Add your .zip here ⬇️

[Slicing_failure.zip](https://github.com/user-attachments/files/16367654/Slicing_failure.zip)

GregValiant, 2024-07-24:
Thanks for the report.
I didn't have any trouble slicing your project file in 5.7.2 on my Win10Pro system.
Immediately after getting a "Won't slice" message use the "Help | Show Configuration Folder" command.  It will take you to the location of the "Cura.log" file.  Zip the log file and post it here.  Maybe it will give a clue as to what is happening on your system.

reuschi, 2024-07-24:
Thanks for your reply and your helping commands. I tried to use your mentioned command and nothing happend (no window opened, or something like that). So, I tried to reinstall Cura 5.7.2 and within the installation process, I selected "Repair". It repaired something (I don't know what exactly) and after reopening and trying to slice, it did everything as expected.
So, from my side this issue can be closed - even if I'm wondering what files could have been missed after the first installation.

Thanks for your willing to help :-) 👍🏻

GregValiant, 2024-07-25:
I'll go ahead and close this then as somehow it seems to be an installation problem.  That is something that might just be system specific.
Good luck.

closed 2024-07-25 by GregValiant