Ultimaker/Cura #19428: slicing failed

wade324, 2024-07-24:
### Cura Version

5.8.0-beta.1

### Operating System

windows 10

### Printer

elegoo 3 plus & creality ender 3 v3 se

### Reproduction steps

updated

### Actual results

will not slice

### Expected results

to slice

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-07-25:
Thanks for the report.
We need the Cura.log file to troubleshoot.  It is in your configuration folder.  Zip it and post the zip folder here.
Have you tried re-installing/repairing Cura?

wade324, 2024-07-25:
Actually I reinstalled it and put my settings back to original settings and it worked fine.  Thanks Wade

GregValiant, 2024-07-25:
I'll go ahead and close this then.