Ultimaker/Cura #19429: Cooling Fan doesn't start at specified height/layer, its always starts at the first layer and stops when it should start cooling

mystermonster94, 2024-07-25:
### Cura Version

5.6

### Operating System

windows 11

### Printer

artillery x1 side winder

### Reproduction steps

download the cura ultimaker 5.8 5.7 or 5.6 i try with those, and its always a problem. u can check it by yourselft if u try to start the fan at hegiht ex: 5mm.. it will start from the first layer.. always DOESNT LISTEN TO THE F start layer height, u can check it looking the G-CODE.. it will be from the first layers always some 106m s1 or more never from the correct height. WTH HAPPEND TO THIS? i was printing and controling fan starting hegiht specific 2 weeks ago, but seems the news updates. are wrong

### Actual results

the actual result its like the slicer isnt hearing the user, when u put the 0.6mm(3th layer) or 0.4mm(2nd layer) start height, it just dosnt working it start ALWAYS at the first layer, and stop at the 2 o 3 depends on the height starting fan, like upsidedown codes bugging...

### Expected results

if i put start at height ex: 3mm it should start the f fan at the 3mm to the top, but ITS LIKE THE F cura slicer is upsidedown, just always starting the fucking fan at 1 layer to the 3mm.... and then stops.. when it should start from the 3mm (15th layer), but it start at the 1 layer and stop at the 15th layer.... so its like some development with the codes from the last version. CANT CONTROL THE HEIGHT TO START FAN.. WHEN IT WAS EASY TO START some crash 1-2 weeks ago.

### Add your .zip and screenshots here ⬇️

[image]

THE 106M is the G-CODE TO PLAY THE FAN AT SOME VELOCITY. WITH THE 106M S(HERES IS 1-255) SPEED

G92 E0
G1 F1500 E-2
;LAYER_COUNT:100
;LAYER:0
M107
G0 F6000 X136.834 Y137.259 Z0.2
G0 X130.018 Y151.243
;TYPE:SKIRT
G1 F1500 E0
G1 F1200 X130.119 Y150.389 E0.03146
G1 X130.257 Y149.506 E0.06416
G1 X130.407 Y148.621 E0.097

(layer 0 is okay)

but when it should start after the selected height or layer ex: 25th layer when it should start .. it start from the 1.. and stops at the 25th layer.. so 

G0 X149.72 Y160.253
;TIME_ELAPSED:83.517869
;LAYER:2
M106 S10.6
;TYPE:WALL-OUTER
;MESH:perno de 8mms.STL
G1 F1500 E52.78735
G1 F1800 X149.782 Y160.442 E52.79463
G1 X149.954 Y160.538 E52.80183

0 X149.005 Y160.872
G0 X149.72 Y160.253
;TIME_ELAPSED:129.198841
;LAYER:4
M106 S21.2
;TYPE:WALL-OUTER
;MESH:perno de 8mms.STL
G1 F1500 E87.79982
G1 F1127.6 X149.782 Y160.442 E87.8071
G1 X149.954 Y160.538 E87.81431
G1 X150.148 Y160.505 E87.82151
G1 X150.259 Y160.331 E87.82906
G1 X150.286 Y160.214 E87.83345
G1 X150.434 Y160.053 E87.84145
G1 X150.854 Y160.003 E87.85693

....

;TIME_ELAPSED:408.943912
;LAYER:24
M106 S127.5
;TYPE:WALL-OUTER
;MESH:perno de 8mms.STL
G1 F1500 E273.1548
G1 F1800 X149.781 Y160.441 E273.16211
G1 X149.954 Y160.538 E273.16936

HERE IT STOPS.. WHEN IT SHOULD START AT THIS LAYER 👍 
G0 X150.143 Y156.06
;TIME_ELAPSED:432.214560
;LAYER:25
;TYPE:WALL-OUTER
;MESH:perno de 8mms.STL
G1 F1500 E288.76784
G1 F600 X150.796 Y155.979 E288.79191
G1 X151.413 Y155.789 E288.81553
G1 X151.982 Y155.502 E288.83884
G1 X152.492 Y155.128 E288.86198

this is driving me bad... im really gona change slicer if this continues.. please someone can solve? i have never this kind a problem, in 5 years using this.. it had some minors bug, but heres is the dam king bug, i unistall 5.8 - 5.6 - 5.7 and 5.0 even... every version has the same fk bug, and i just delet the profiles already and make all of the version from fabric, it still doesnt woking... if u can try this u willl see the bug, that wasnt there a few days ago.. or im the only bad lucky guy

Asterchades, 2024-07-25:
Those commands are correct, though. You're seeing a build up through 10.6 (approximately 4%) on layer 3, 21.2 (approximately 8%) on layer 5, up to 127.5 (50%) on layer 25, which corresponds to the 25-layer build-up you've requested. Layer 26 then doesn't have a command at all to change the speed, so no further changes should be getting made at that point, meaning any changes to the actual speed in use are outside of Cura's control. *Note that the GCode layer comments are zero based, meaning the first layer is layer 0 rather than layer 1).*

This, so near as I can figure, is likely to indicate an electrical fault on your control board. Specifically the MOSFET responsible for powering the fan is undergoing failure, but hasn't completely failed *yet*. I'm taken to believe it's called an "Avalanche failure" but that's not something I can address any further with my knowledge of how they work.

*If I'm correct* you'll see exactly the same result with a different slicer - *any* slicer. The situation is also unlikely to resolve itself as the MOSFET will eventually fail. You just have to hope that when it does it's failure mode doesn't send 24v to the MCU, else it's going to have a very bad day. MOSFETs will *typically* fail open but they don't have to.

If the images I can find are to be believed, on the factory mainboard the fan MOSFET is going to be Q3, Q4, or Q5 depending on if you're using FAN0, FAN1, or FAN2 respectively.

GregValiant, 2024-07-25:
There is a new post processor "Advanced Cooling Fan Control".  You can give that a try.  It provides much more control of the fan speed and works either "By Layer" or "By Feature".
"By Layer" is better for small prints because of the lag time of the fan spooling up and slowing down.
Enter the setting as Layer#/Fan Speed %.  In this example I've set the fan to turn on on layer 5 (gcode layer:4) at 30% duty cycle.  At layer 20 it will go to 50% duty cycle, and at layer 30 it will turn off.
[image]
The gcode lines would be:
;LAYER:4
M106 S77
;LAYER:19
M106 S127
;LAYER:29
M106 S0

For large prints where the fan has time to get to a particular speed "By Feature" can be a better choice.
[image]

If you have a hardware problem then that's a different story.  As Asterchades points out, at LAYER:24 (Cura preview layer 25) the fan line "M106 S127.5" has set the fan to 50% duty cycle.  The fan will (is supposed to) stay at that speed until another M106 changes the speed, or an M107 shuts the fan off.

_"that wasnt there a few days ago.."_
That could be a clue.

mystermonster94, 2024-07-25:
> Those commands are correct, though. You're seeing a build up through 10.6 (approximately 4%) on layer 3, 21.2 (approximately 8%) on layer 5, up to 127.5 (50%) on layer 25, which corresponds to the 25-layer build-up you've requested. Layer 26 then doesn't have a command at all to change the speed, so no further changes should be getting made at that point, meaning any changes to the actual speed in use are outside of Cura's control. _Note that the GCode layer comments are zero based, meaning the first layer is layer 0 rather than layer 1._
> 
> This, so near as I can figure, is likely to indicate an electrical fault on your control board. Specifically the MOSFET responsible for powering the fan is undergoing failure, but hasn't completely failed _yet_. I'm taken to believe it's called an "Avalanche failure" but that's not something I can address any further with my knowledge of how they work.
> 
> _If I'm correct_ you'll see exactly the same result with a different slicer - _any_ slicer. The situation is also unlikely to resolve itself as the MOSFET will eventually fail. You just have to hope that when it does it's failure mode doesn't send 24v to the MCU, else it's going to have a very bad day. MOSFETs will _typically_ fail open but they don't have to.
> 
> If the images I can find are to be believed, on the factory control board the fan MOSFET is going to be Q3, Q4, or Q5 depending on if you're using FAN0, FAN1, or FAN2 respectively.

Thanks for answering and trying to figured out the problem but it isnt a hardware problem, or mosfets, cuz as you see the codes works alright when the printer reads the 106m s(speed) it works at the % that you said… i was meaning that 1 week ago, it was hearing the layer height to start the fan after the height correctly…  i didnt mean to use ramps for just for starting the fan one time after a specific height… it just shouldnt start before the height or layer and always from the first Layer… its like a upsidedown code.. but maybe the option isnt avaliable anymore, i dont neee to change diferent speeds.. just 50% speed is okay. The problema is the time when it should start, why starting always at first layer of the layer height when layer height should rule the beggining of the fan to start spining. Its a development problem. Otherwise it should give 0 m106 g-code and only should appear when the printer reach the layer height correctly and start the fan… this basic feature that was working and now it isnt.. by some bug problem. I mean the layer height it isnt hear by the cura.. thanks anyway

Asterchades, 2024-07-25:
Cura hasn't waited to start the fan at a specific layer as far back as 4.0, which released over five years ago (I can't speak to earlier as I never used it). It's always ramped up from the **Initial Fan Speed** on layer 1 to the **Regular Fan Speed** for the layer corresponding to the **Regular Fan Speed at Layer**. Even the tooltip for the Initial Fan Speed indicates this ramp is going to happen:

> The speed at which the fans spin at the start of the print. In subsequent layers the fan speed is gradually increased up to the layer corresponding to Regular Fan Speed at Height.

The only way for it to have been any different would be to set the fan speed settings to 0% and then either rely on the **Maximum Fan Speed** kicking in, or to use the ChangeAtZ post-processor to explicitly make the change at your requested layer. Or, more recently, the script that Greg has referred to above.

I do know that Slic3r and its derivatives (including SuperSlicer, PrusaSlicer, and Orca Slicer) have a setting to completely disable the fan for a user-specified number of layers, but Cura hasn't had that as a direct option for at least five years (if it ever had it).

Beyond this I'm having difficulty determining what exactly is backwards, upside-down, or stopping, which is making it hard to offer any additional feedback. I feel like you're using those terms unnecessarily, obfuscating the issue.

GregValiant, 2024-07-25:
@mystermonster94  I bought my printer (Ender 3 Pro) about 4 years ago and the 4010 layer cooling blower was so bad that the Cura fan settings just didn't matter.  The fan was weak, and the little air it moved was aimed completely wrong.  Fortunately for me it was such a complete piece of crap that it died after only a couple of months.

I replaced it with a 5015 ball bearing unit that actually worked and the duct I designed put the air exactly where it needs to be.  Now that the fan worked, it became apparent that Cura did not offer what I wanted in fan control.  I print mostly functional items in PETG and I wanted to be able to turn the fan off except in certain places like overhangs, first skin over infill, support-interfaces, etc..  Cura does not have the ability to turn the fan off mid-print.  It just starts at 0, and increases up to the set speed at some layer, and then shuts off when the print is finished.  Some of the "early layer" M106 lines that Cura inserts are dumb since my fan doesn't start until 12% which would be M106 S31.  A setting like M106 S10.5 (4%) just doesn't work at all.

I went back in time through the Feature Requests and found that better fan control had been requested since at least 2018 and it had never been addressed.  I figured if it was going to happen I was going to have to do it.  So I started on that post processor.  Since I knew zip-zilch about Python (some indelicate persons would say I still don't) it was a struggle, but it finally got done.

"Advanced Cooling Fan Control" is extremely flexible.  It allows fine control of the fan for pretty much any situation.  You can run multiple instances of the post processor and change from "By Layer" to "By Feature" and back again.  It will control up to 4 fan circuits (P0 to P3).  You can even choose to cool the raft and turn the fan off during long travel moves.  For some prints you need to take a close look at the Cura preview and take some time to build your fan profile for the specific print.   It's what I needed, it's what I wanted, and it's what I use.  For PLA prints I start with the 5/30 setting (that's why it's the default) and then increase to 20/100.  You could choose to do something like:
5/50
10/0
20/100
30/0
That would start at _exactly_ layer 5 and 50% fan, turn the fan off at layer 10, turn back on at layer 20 at 100%, and turn off again at layer 30.

Flexible means complex.  It is up to the user to configure it so it provides a proper cooling profile for a particular print.  The "first instance" of the script should always have "Remove M106 lines prior to inserting new" checked so it deletes all the M106 lines that Cura inserted.  Any subsequent instances must have that setting disabled or it will wipe out the M106 lines that the first instance added.

So Cura now has the capability of complete fan control.  I'm inclined to close this as I just don't see a problem provided you use the post processor.  It was first included with (I think) 5.7.1 but it will work with 5.6.0 with no problem as that was what I used for most of the development.  You can download it from the Cura code pages under [plugins/postprocessingplugins/scripts](https://github.com/Ultimaker/Cura/tree/main/plugins/PostProcessingPlugin/scripts).  It would go into your Configuration Folder and the "scripts" sub-folder.
All of us are free to use any slicer we like.  UM Cura is not the only slicer I use, but it is my favorite.

mystermonster94, 2024-07-26:
> @mystermonster94 I bought my printer (Ender 3 Pro) about 4 years ago and the 4010 layer cooling blower was so bad that the Cura fan settings just didn't matter. The fan was weak, and the little air [...]

mine too, after i read you reply i think i ll give a chance to the post procesator fan control from cura. thanks for reply

closed 2024-07-26 by GregValiant