Ultimaker/Cura #19431: Wont Slice Articulated

Dicealope, 2024-07-25:
### Cura Version

5.8

### Operating System

Windows 11

### Printer

Anycubic Predator

### Describe your model

I've printed articulated before, but that was a few updates ago.

### Add your .zip here ⬇️

[AP_butterfly dragon f (1).zip](https://github.com/user-attachments/files/16381918/AP_butterfly.dragon.f.1.zip)

GregValiant, 2024-07-25:
Thanks for the report.
I used Cura Mesh Tools to check the model.
Check summary:
- butterfly dragon f (1).stl is not watertight and may not print properly and consists of 38 submeshes

This is a full report:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
**--> 6 Non-manifold edges (?)
--> 10 Inverted faces (?)**
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 41193ms ------
-> Vertex count changed from 219225 to 219772 (+547)
-> Triangle count changed from 438610 to 440258 (+1648)
Those might be minor errors or the "Non-manifold edges" might be wide open enough that Cura is becoming confused about where the outside ends and the inside begins as it slices across an opening in the mesh.

This is the repaired model.  It took 18.8 minutes to slice at your 0.12 layer height and 0.20 line width.  As you can see this is a 42 hour print.
[image]

I'll leave the bug label on this for now but it looks to be (at least partly) a model problem.  Let us know if repairing the model fixes the problem for you.
FIY the worst model I've seen around here also happened to be an articulated model called a "Crystal Dragon" and it had over 30,000 errors in it.  It wouldn't slice and it was too bad to repair.

LilBub, 2024-07-29:
@GregValiant I've printed a lot of articulated models that alert of an issue but slice/print fine.  
How are you repairing them?  Inside Mesh Tools?

closed 2024-08-09 by github-actions[bot]

GregValiant, 2024-08-09:
@LilBub I use Microsoft 3D Builder for my own repairs.  I use [Formware.co](https://formware.co/OnlineStlRepair) to generate that fancy report.  That online site is free but has a file size limit and a 4 minute timeout and it's often quite backed up.  I like the report though.  I'm looking forward to coming across my first "Degenerate Face".  I'm wondering if it will be a photo of my ex-wife.