Ultimaker/Cura #19432: Error while trying to Initialize the Virtual Python Development Environment

AbdElhalim12, 2024-07-25:
### Cura Version

5.7.2

### Operating System

windows 11

### Printer

none

### Reproduction steps

I'm trying to run cura form the source, I got this error while trying to Initialize the Virtual Python Development Environment

### Actual results

Configuration (profile_host):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.cppstd=17
compiler.runtime=MD
compiler.version=17
os=Windows
os_build=Windows
curaengine:compiler.cppstd=20
curaengine*:compiler.cppstd=20
curaengine_plugin_infill_generate:compiler.cppstd=20
curaengine_plugin_gradual_flow:compiler.cppstd=20
curaengine_grpc_definitions:compiler.cppstd=20
scripta:compiler.cppstd=20
umspatial*:compiler.cppstd=20
dulcificum:compiler.cppstd=20
cura_settings:compiler.cppstd=20
[options]
cura:devtools=True
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True
tools.build:skip_test=False

Configuration (profile_build):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.cppstd=17
compiler.runtime=MD
compiler.version=17
os=Windows
os_build=Windows
[options]
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True
tools.build:skip_test=False

WARN: pyarcus/5.3.1: requirement protobuf/3.21.9 overridden by uranium/5.8.0-beta.1 to protobuf/3.21.12
WARN: pyarcus/5.3.1: requirement cpython/3.10.4 overridden by uranium/5.8.0-beta.1 to cpython/3.10.4@ultimaker/stable
[... 276 lines cut ...]
cpython/3.10.4@ultimaker/stable: WARN: Build folder is dirty, removing it: C:\.conan\9d9e14\1
cpython/3.10.4@ultimaker/stable: Copying sources to build folder
cpython/3.10.4@ultimaker/stable: Building your package in C:\.conan\9d9e14\1
cpython/3.10.4@ultimaker/stable: Generator 'MSBuildDeps' calling 'generate()'
cpython/3.10.4@ultimaker/stable: Generator txt created conanbuildinfo.txt
cpython/3.10.4@ultimaker/stable: Aggregating env generators
cpython/3.10.4@ultimaker/stable: Calling build()
cpython/3.10.4@ultimaker/stable: Patching runtime
cpython/3.10.4@ultimaker/stable:
cpython/3.10.4@ultimaker/stable: ERROR: Package 'c4d147f2014dded40c02392443a406fea3df78e1' build failed
cpython/3.10.4@ultimaker/stable: WARN: Build folder C:\.conan\9d9e14\1
ERROR: cpython/3.10.4@ultimaker/stable: Error in build() method, line 425
        self._msvc_build()
while calling '_msvc_build', line 394
        msbuild = MSBuild(self)
        ConanException: Using the wrong 'MSBuild' helper. To use MSBuildDeps, MSBuildToolchain you should use 'from conan.tools.microsoft import MSBuild'. Set environment variable CONAN_DISABLE_STRICT_MODE=1 to override this check (should only be used to build old packages).

### Expected results

not getting error.

### Add your .zip and screenshots here ⬇️

[image]
[image]

Antier, 2024-08-13:
I am also facing the same problem. Do you have a solution?

Antier, 2024-08-13:
i have already dealt with it
edit file : ~/.conan/profiles/default.  

[env]
CONAN_DISABLE_STRICT_MODE=1

AbdElhalim12, 2024-08-23:
I'm getting this error 
ERROR: curaengine/5.9.0-alpha.0+1b78d4@ultimaker/testing: Cannot load recipe.
Error loading conanfile at 'C:\Users\AbdElhaleem\.conan\data\curaengine\5.9.0-alpha.0+1b78d4\ultimaker\testing\export\conanfile.py': Unable to load conanfile in C:\Users\AbdElhaleem\.conan\data\curaengine\5.9.0-alpha.0+1b78d4\ultimaker\testing\export\conanfile.py
  File "c:\users\abdelhaleem\appdata\local\programs\python\python39-32\lib\imp.py", line 171, in load_source
    module = _load(spec)
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 786, in exec_module
  File "<frozen importlib._bootstrap_external>", line 923, in get_code
  File "<frozen importlib._bootstrap_external>", line 853, in source_to_code
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "C:\Users\AbdElhaleem\.conan\data\curaengine\5.9.0-alpha.0+1b78d4\ultimaker\testing\export\conanfile.py", line 226
    match self.settings.os:
          ^
SyntaxError: invalid syntax

PS C:\Users\AbdElhaleem\cura>

while running conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv -g PyCharmRunEnv