Ultimaker/Cura #19433: Turn off printer name on saving file does not work

donaldv9, 2024-07-25:
### Cura Version

7.7.1

### Operating System

windows 10

### Printer

Elegoo Neptune 3 pro

### Reproduction steps

1. Unchecked; add machine prefix to job name

### Actual results

saved file and it still added the machine prefix to the job name even though it does not show on the bottom left of screen.

### Expected results

Saving the file should not have the machine prefix added to the job name when it is unchecked in preferences.

### Add your .zip and screenshots here ⬇️

[image]
[image]

donaldv9, 2024-07-25:
Made an error on the version number. It is 5.7.1

donaldv9, 2024-07-25:
[image]

GregValiant, 2024-07-25:
Thanks for the report.
I can't duplicate this.  I have the same printer and I tried with 5.7.1 on a Win10 system.  It's a setting I always disable and I've never had a problem with it.
The Cura.log file is in your Configuration Folder.  Zip it and post it.  It might have a clue as to what is stuck on your system.

donaldv9, 2024-07-26:
Thank you Greg for your help. I have the Cura.log file attached in zip format as you requested.

Donald Velasco

GregValiant, 2024-07-26:
Sorry but attachments don't work via email.  They must be posted directly to the site (Github rules).

donaldv9, 2024-07-26:
[curalog.zip](https://github.com/user-attachments/files/16395505/curalog.zip)

Sorry Greg, here it is.

GregValiant, 2024-07-26:
We're having a heck of a time here.  The zip folder is empty.

donaldv9, 2024-07-26:
[cura.zip](https://github.com/user-attachments/files/16397444/cura.zip)

on my end it shows the size as 273kb in zipped format. not sure what happened the first time but this should work. Thanks Greg!

GregValiant, 2024-07-27:
Got it.
There are a lot of warnings that I'm not familiar with.  You have several plugins installed as well.  Maybe a plugin like "MakerBotWriter" didn't receive the setting change regarding the file name prepend.

@HellAholic of the Cura team will take a look.

donaldv9, 2024-07-27:
Thank you so much Greg and Hellaholic for your help persistence on getting this resolved. I have removed the plugin and it is now working perfectly! You all are an amazing team and I am so grateful for all your help!

donaldv9, 2024-07-27:
[image]
showing that it is fixed.

GregValiant, 2024-07-27:
Thanks guys.  I'll go ahead and close this.

closed 2024-07-27 by GregValiant