Ultimaker/Cura #19437: Cura 5.7.2 inserts M104 and M109 before start code even when M104/M109 S{material_print_temperature_layer_0} is in start code.

Austenite2, 2024-07-26:
### Cura Version

5.7.2

### Operating System

Windows 11

### Printer

Voron 2.4

### Reproduction steps

In Machine Settings, Start G-code, add:
```
M140 S{material_bed_temperature_layer_0}
M104 S{material_print_temperature_layer_0}
M190 S{material_bed_temperature_layer_0}
M109 S{material_print_temperature_layer_0}
print_start
```

Also happens if use M104 S{material_print_temperature} instead

### Actual results

Generated G-code still has M104 and M109 commands:
```
;Generated with Cura_SteamEngine 5.7.2
T0
M104 S265
M109 S265
M82 ;absolute extrusion mode
;Filament name = Progressive3D ABS Dark Grey Flow 90.5/91
;Filament type = ABS
;Filament weight = [85.19217265625001]
;Nozzle diameter = 0.6
M140 S125
M104 S265
M190 S125
M109 S265
print_start
```

### Expected results

M104 and M109 lines aftert the T0 line should not be added to G-code.

Seems similar to #10

### Add your .zip and screenshots here ⬇️

[image]

LilBub, 2024-07-27:
I could be wrong, seems similar to https://github.com/Ultimaker/Cura/issues/19190

GregValiant, 2024-07-27:
Yeah, this is a duplicate of #19204 .
@Austenite2 This bug has been fixed in 5.8beta so I'll mark this as a duplicate and go ahead and close it.

I've written a post-processor for 5.7.2 that removes those lines that Cura adds.
Unzip [CuraPrependBugFix.zip](https://github.com/user-attachments/files/15806103/CuraPrependBugFix.zip) and put the ".py" file into your configuration folder and the "scripts" sub-folder.
The post-processor will be available with the rest of them under "Extensions | Post Processing | Modify Gcode" and then "Add Script".

closed 2024-07-27 by GregValiant

Austenite2, 2024-07-27:
Apologies for the duplicate, and thanks for the script!