Ultimaker/Cura #19438: Improve Japanese localization

h1data, 2024-07-27:
### Cura Version

5.7.2

### Operating System

not specific

### Printer

not specific

### Reproduction steps

Hello, I've recently considered improving Cura's Japanese localization.
(and discovered issue #19401)

Based on the [Translating Cura](https://github.com/Ultimaker/Cura/wiki/Translating-Cura) wiki page, I have been preparing PR for 5.7.2, but other translation words were added in the 5.8 branch.
Also, it is said your team uses Lokalise translation service in #18614.

How do I join the translation work?

### Actual results

There are problems in Japanese localization.

### 1. Mistranslations

#### Example
```
msgctxt "@action:inmenu"
msgid "Toggle Full Screen"
msgstr "留め金 フルスクリーン"
```

"留め金" might be from "Toggle", but rather a metal hook that holds doors/gates.<br> 
It seems due to bad machine translation and not checked in the actual program.<br>
(it would be "全画面表示切替")

That is just an example and there are other mistranslations.

### 2. Inconsistent terms

The common terms in the program should be consistent but there are some inconsistent translations.

For example, "material" is a common term but there are different translations; "フィラメント" (filament), "材料" (translated straightly), or "マテリアル" (katakana for "material").

### 3. misuses of `&` operators for the menu shortcut keys.

`&x` indicates the shortcut key `x` in menus, however, there are misuses for `&` operators.

#### Example
```
msgctxt "@title:menu menubar:toplevel"
msgid "&File"
msgstr "&ファイル"
```
`&フ` is not recognized as a shortcut key, so just displayed as "&ファイル".

In most Japanese software, the menu shortcut key is displayed in parenthesis like "ファイル(<ins>F</ins>)", thus it would be "ファイル(&F)". 

### 4. Multiline is not displayed properly

#### Example
```
msgctxt "@text"
msgid ""
"- Add material profiles and plug-ins from the Marketplace\n"
"- Back-up and sync your material profiles and plug-ins\n"
"- Share ideas and get help from 48,000+ users in the UltiMaker community"
msgstr "- マーケットプレースから材料プロファイルとプラグインを追加- 材料プロファイルとプラグインのバックアップと同期- UltiMakerコミュニティで48,000人以上のユーザーとアイデアを共有してアドバイスをもらう"
```
Those lines are not broken into lines. (see additional information)

There should be `\n` at each end and split into lines like below.

```
msgctxt "@text"
msgid ""
"- Add material profiles and plug-ins from the Marketplace\n"
"- Back-up and sync your material profiles and plug-ins\n"
"- Share ideas and get help from 48,000+ users in the UltiMaker community"
msgstr "- マーケットプレースから材料プロファイルとプラグインを追加\n"
"- 材料プロファイルとプラグインのバックアップと同期\n"
"- UltiMakerコミュニティで48,000人以上のユーザーとアイデアを共有してアドバイスをもらう"
```

### Expected results

see actual results

### Add your .zip and screenshots here ⬇️

example for 4. Multiline is not displayed properly
[image]

h1data, 2024-07-27:
Thanks for the invitation!
I would like to receive emails for beta branches.
Please tell me the email address to send it, or DM me on X (`@h1data`).

I will make a PR to the main branch for future releases.

h1data, 2024-10-12:
@wawanbreton Thanks for merging the PR!

I found untranslated lines in the 5.9 branch
and translation suggestions by a Japanese user.
https://note.com/y_labo/n/nee490a276494
At this moment, is 5.9 the branch to make PR?

Btw, please check the [PR](https://github.com/Ultimaker/Uranium/pull/960) in the Uranium repository too.

wawanbreton, 2024-10-14:
> I found untranslated lines in the 5.9 branch and translation suggestions by a Japanese user. https://note.com/y_labo/n/nee490a276494 At this moment, is 5.9 the branch to make PR?

Right now, I have sent the labels to our translation service, and they are ready to be integrated. This means thah quite soon, the 5.9 branch will be updated and there should be no more missing translations. No need to do the translation then, but you can still make a review on the new labels.

> Btw, please check the [PR](https://github.com/Ultimaker/Uranium/pull/960) in the Uranium repository too.

Oh sorry I forgot that. I hope that will not conflict with the fresh new translations :grimacing: 

I also see on the discussion that you were willing to register for regular Japanese translation, is that still the case ? We currently do it via a professional service, but having them done by a contributor has many advantages.

h1data, 2024-10-15:
Yes, I'm willing to contribute to the product!
However, to guarantee the quality, it is better to have others double-check as mentioned in #19473.
I will consider the translation process after the translation update in 5.9.
(when should I translate, if the service is necessary, etc)

wawanbreton, 2024-10-16:
> Yes, I'm willing to contribute to the product!

Cool, let's go for it :slightly_smiling_face: what is the easiest way I can get your email ?

> when should I translate

When we do the feature freeze for a new release, we create the feature branch then immediatly update the translations files. Then we send the new labels to our translations service, and for contributed languages we send an email to the translators. That usually lets a few weeks for them to make a PR with the updated files. On average it is about 40 new labels, every ~3 month.

> if the service is necessary

We usually trust the contributors to do it correctly :slightly_smiling_face: although professional translators may have more experience, they usually miss the context and the usual wording, especially in 3D printing which is very technical. However if you have some doubts, just tell us and we will have the translations double-checked.

h1data, 2024-10-16:
I have a contact form instead of a public address.
Could you contact me via the URL below? I will reply to the input address.
https://h1data.github.io/contact.html
Or DM `@h1data` on X if you have an account.

h1data, 2024-10-28:
@wawanbreton 
Thanks again for receiving and merging the updates!
We have accomplished to solve those issues above for now.
I'll close this issue, however, there is still a concern
about applying our changes to the translation service.
I would like to discuss it in future updates.

closed 2024-10-28 by h1data

wawanbreton, 2024-10-28:
> I would like to discuss it in future updates.

Well, for future updates I will not send the Japanese translation to Lokalize, only to you, so there should be no confusion. In case you are too busy, we can still send them to Lokalize later because they are usually very fast. Did you have something different in mind ?

h1data, 2024-10-28:
I think it is the straight way, too.
I can translate for usual updates in at least a few days.
Also, I would like to find others who can translate or proofread for quality.

bgpat, 2024-10-28:
> Also, I would like to find others who can translate or proofread for quality.

I can review Japanese translations. Feel free to contact me if needed.

h1data, 2024-10-28:
@bgpat 
Thank you for reaching us!
I would like to have the translations reviewed.
I'll send details on DM (X).

@wawanbreton 
No person with 3D printers around me, but Tanaka-san offers help!
I'll be glad if he helps to check the translations.

wawanbreton, 2024-10-28:
Awesome, thanks @bgpat for the help :slightly_smiling_face:

PR #19473 (2024-10-08): improve Japanese localization
# Description

Improving Japanese localization 
Resolves: #19438

## Type of change

- [x] Translations

# How Has This Been Tested?

- [x] Compiled "po" files to "mo"  and checked the view of the application.

**Test Configuration**:
* Operating System: Windows 10 

# Checklist:

- [x] My code follows the style guidelines of this project as described in [UltiMaker Meta](https://github.com/Ultimaker/Meta) and [Cura QML best practices](https://github.com/Ultimaker/Cura/wiki/QML-Best-Practices)
- [x] I have read the [Contribution guide](https://github.com/Ultimaker/ [... cut]
PR #19817 (2024-10-28): improve Japanese translation for 5.9
# Description

Improving Japanese localization for 5.9
Resolves: #19438

## Type of change

- [x] Translations

# How Has This Been Tested?

- [x] Compiled *.mo file by gettext and checked the appearance of the built application.

**Test Configuration**:
* Operating System: Windows 11

# Checklist:

- [x] I have read the [Contribution guide](https://github.com/Ultimaker/Cura/blob/main/CONTRIBUTING.md)