Ultimaker/Cura #19440: Ender 3 S1 Pro change color

omniderb, 2024-07-28:
### Is your feature request related to a problem?

The Cura ‘Post Processing/Modify gcode’ does not work with my new Ender 3 S1 Pro.  

### Describe the solution you'd like

Please provide a gcode that will work with the S1 Pro.

### Describe alternatives you've considered

I contacted Creality. They suggested that I ask you.

### Affected users and/or printers

I only aware of the effect on the Ender 3 S1 Pro.

### Additional information & file uploads

[image]

gh13, 2024-07-28:
Are you printing from the SD Card?

omniderb, 2024-07-28:
Yes. I am printing from the SD card.

gh13, 2024-07-28:
That's OK as it wouldn't work if you were printing via USB
This is my S1 Pro pause configuration, which works on my printer. The script I use is an updated version that the dev posted somewhere, which adds a couple more parameters & the ability to disable it when not needed. but when I was using the original V1 version like you the comparable settings worked & your setting seem like they should work. I've attached the updated version if you want to try that
FWIW The S1 Pro doesn't support the beep & LCD message so don't know if that's affecting anything

[image]
[PauseAtLayer.zip](https://github.com/user-attachments/files/16405034/PauseAtLayer.zip)

omniderb, 2024-07-28:
Thank you.  How do I put that file into Cura?  Sorry, I’m not very tech
savvy.

gh13, 2024-07-29:
in cura go to help > show configuration folder,  copy the .py  file to the scripts folde & restart cura. you should then be able to add the post processor