Ultimaker/Cura #19441: Missing infill in small parts

Schatta123, 2024-07-28:
### Cura Version

5.8.0-beta.1

### Operating System

Windows 11

### Printer

Biqu B1

### Reproduction steps

In a object smaller than 3mm Cura 5.8 beta 1 does not put infill like in this picture:
[image]
While in cura 5.7 it does put infill. Here is the same example:
[image]

### Actual results

When this happen I do not get infill in those parts

### Expected results

It should have infill in that part

### Add your .zip and screenshots here ⬇️

I have already attached the screenshots

GregValiant, 2024-07-29:
Thanks for the report.
The brown lines in both images look like Infill.  Either the images aren't describing the problem, or I'm just not getting it.
Put some arrows on the images to point to where you see a problem.  A project file would be helpful.
When you choose an infill that "moves" from layer-to-layer (ex: "cubic"), and the part is small, then some areas will appear to be missing infill because it has moved away.  If the part was taller you can see it come back.  The infill density also plays a part.

Schatta123, 2024-07-29:
Here it is suppose to be infill but cura does not put it (Cura 5.8 Beta 1)
[image]
And if I use the same profile but in cura 5.7 all work fine and there it has infill
[image]
It only happen when the object length is less than 3mm. Here I made another example in cura 5.8 Beta 1
[image]
As you can see the object is hollow but it should have a 30% cubic infill. So it is not working as it should.
For now I will use Cura 5.7 until this bug has been solve. Sorry I do not know how to send a project file. I hope now is more clear. Thank you