Ultimaker/Cura #19442: Floating tree support (Cura 5.8 Beta 1)

LilBub, 2024-07-29:
### Cura Version

5.8.0 Beta 1

### Operating System

Windows 11

### Printer

Mingda Magician Max2 and CR-10S

### Reproduction steps

1) Slice model, enabled brim and tree supports

### Actual results

Rear tree support is created about an inch above the build plate.  That would result in a nasty fail.

[image]

### Expected results

Tree Support should be on the build plate

### Add your .zip and screenshots here ⬇️

Project file added.
[MMMAX_C3PO_Head_Front_-_Without_Tab_-_JesseM_-_2-1-2023.zip](https://github.com/user-attachments/files/16405715/MMMAX_C3PO_Head_Front_-_Without_Tab_-_JesseM_-_2-1-2023.zip)

GregValiant, 2024-07-29:
Thanks for the report.
That's really bad and I was able to duplicate the problem.
[image]

@HellAholic of the Cura team will take a look.

LilBub, 2024-07-31:
This seems appropriate to say, given this affected C-3PO - "Thank the Maker!"   :)

MariMakes, 2024-08-01:
👋 Quick update on our side.
This is resolved in the 5.8 release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.8.0](https://github.com/Ultimaker/Cura/releases/tag/5.8.0)

I'll close this issue since it's resolved.
Thanks again, and please let us know if you run into any other issues 💪

closed 2024-08-01 by MariMakes

LilBub, 2024-08-01:
@MariMakes thanks, I see the website still shows 5.7.2 tho?
Or are we still Beta?

[image]