Ultimaker/Cura #19444: Cura ignores Retraction and Z Hop Settings

jseiffert14, 2024-07-29:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

RatRig V-Core 3 (Klipper)

### Reproduction steps

Hi,
I want to print a part with two different extruders.
In the Travel settings, I set the Retraction Distance and Z-Hop for each extruder (see pictures attached).
Extruder 4 (T3) is the main extruder. Extruder 3 (T2) is the second extruder used.
When I slice the part, Cura outputs a G-Code file, but **always** uses the retraction settings for the main extruder used (see G-Code file attached).
[image]
[image]
[image]

### Actual results

Cura ignores the extruder specific settings and always uses the settings for the "main" extruder, which was selected to print the part. If features (like infill or walls) are printed with a different extruder, Cura still uses the settings of the first/main extruder. 
That's actually quite dangerous, as wrong settings can lead to damage and malfunction.
I attached all the pictures, G-Code file and project file in the *.zip file.

### Expected results

Cura should use the settings, which can be set in the respective extruder-specific settings.
I don't understand why it is possible to adjust all the settings for each extruder and then they're ignored. Am I doing something wrong ?

### Add your .zip and screenshots here ⬇️

[Cura-RetractionBug.zip](https://github.com/user-attachments/files/16410381/Cura-RetractionBug.zip)

GregValiant, 2024-07-29:
Thanks for the report.
This looks like a bug to me as well.  Changing the project to involve T0 and T1 yields the same erroneous results.  The same condition exists with a UM S7 as the active printer.

@HellAholic and the Cura team will take a look.

jseiffert14, 2024-07-30:
Hi,

thanks for the quick answers.
Here's the correct G-Code file and the corresponding project file. Sorry for uploading a G-Code file sliced with a slightly different profile.
@HellAholic you're right, I showed a wrong part of the G-Code. The bug is still there though. I attached a better screenshot below. It seems that the nozzle change retraction distance is used according to each extruder, but the travel retraction settings are ignored. Hope that helps to clarify the issue here.

![G-Code retraction_T2](https://github.com/user-attach
[Cura BugV2.zip](https://github.com/user-attachments/files/16424107/Cura.BugV2.zip)
ments/assets/039629c1-2aef-405e-8420-3f16a2b15bcf)

closed 2025-01-10 by HellAholic

HellAholic, 2025-01-10:
With the change in the linked PR, the issue should be resolved.
For retractions, now each extruder setting is taken into account.
Please double check this in the beta period for the next release to see if it works as you would expect or there are areas that still need attention. If you raise the issue during beta period, there is a higher chance of the fix being included in stable rather than waiting for the next release.

Cheers!