Ultimaker/Cura #19452: Broken preview at Cura 5.7.2

NestlyS, 2024-07-30:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

Flying Bear Ghost 6

### Reproduction steps

1. I deleted some old extention. I don't know the name.

### Actual results

Sliced preview looks very strange
[image]

Also there is no layer slider at the bottom
[image]

Only top layer looks ok
[image]

I make some research about that problem and it looks like Cura can't find acceptable videocard, but I have Nvidia 3090 inside. 

### Expected results

There's no bugs in model preview.

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]

Asterchades, 2024-07-31:
This is what the Compatibility Mode view looks like, as indicated by the dropdown in the upper right corner of your third and sixth screenshot. Nothing here is untoward with regards to what you're seeing.

As to why it's using it? There's two reasons:

The first is that you've manually enabled it - there's a setting called, rather plainly, "Force layer view compatibility mode (restart required)" which will, as the name suggests, force its use. Checking or unchecking this will toggle the view in subsequent loads of the Cura client.

The second is that Cura cannot find an appropriate OpenGL device. It's looking for the presence of both OpenGL 4.1 (or better) hardware _and_ software - both are required to use it. **Without the log requested when you first opened this issue** there's no way to know which it is unable to find. Note that driver updates from Microsoft (via Windows Update) tend to break OpenGL functionality, and laptops can also encounter issues with regards to switchable graphics (though I'm unaware of any mobile 3090 hardware, and mention this only for posterity).

Should you not have toggled the setting I would encourage you to reinstall your graphics drivers, and ensure your primary monitor is connected to the discrete graphics card. If that doesn't resolve the situation then more information is going to be needed - such as the Cura.log file you will find in the configuration folder (use the help menu item to quickly navigate there).

NestlyS, 2024-07-31:
@Asterchades , I don't know why  "Force layer view compatibility mode (restart required)" was enabled, but it was. Thanks, after toggling it off everything is acts fine!

closed 2024-07-31 by NestlyS