Ultimaker/Cura #19453: Lower/Higher/Different Temperature for upper support/lower support connected model layers

jacc190676, 2024-07-31:
### Is your feature request related to a problem?

Removing Support (specially for big flat connections) werry difficult even with all known "cheats" (without special support filament)

### Describe the solution you'd like

Maybe a change of nozzle temperature for the highest suport layers (and/or interface/bottom/roof) (eg. lower Temp or over temp)
and maybe in first layers wich from model connect to support, could lead to make support adhesion to model less strong to remove more easy. We know from temp towers, that different temperatures lead to less strong layer connections, maybe a lower temperature to connect support/model could help. we can think about lowering OR incrasing temperature (lower = less adhesion / higher=filament "burnout" to use it outside of its recommended settings to damage its structure and make it more volunerable to brick it?

\_\/_\/_\/_\Model/_\/_\/_/   <- Model ( _\/_  lowest layer(s) other temperature)
^^^^^\/Model\/^^^^^   <-Support (^ upper layer(s )other temperature)
^^^^^\/Model\/^^^^^

### Describe alternatives you've considered

Tried many tutorials for optimizing less support adhesion to model, but if support is a "big flat" thing, e.g. if support is a flat big table on a tower, then removing this big flat support is verry difficult

### Affected users and/or printers

Everyone without special support filament

### Additional information & file uploads

-

GregValiant, 2024-07-31:
Changing the temperature always takes a while and the nozzle is subject to oozing when it isn't extruding.  I've tried a couple of different methods to try and fix the problem of support sticking too well to the model (like having a "fan only" pass just above the interface was one of them) but in the end I decided that changing filament was the way to go.
I have written "Support Interface Material Change" specifically for "large flat areas".  If the model is being printed in PLA then you can switch filament to PETG for the top layer of the interface and then switch back to PLA when the model resumes.

That is also the main downside because if you over-use the script there can be a LOT of pauses.  If the area of the Interface happens to be small, then the pauses will be bang-bang and occur in quick succession.  That's OK but...nervous.

1. Two pauses are required on the layers you indicate are "of interest".  That can mean a lot of pauses but for a single large flat Support-Interface, the script works great and requires just two pauses.
2. Your printer must be capable of pausing.  "M0" works on my printer and I included other pause commands in a drop-down list.  M600 can be used as a "Custom" command, but it requires that you enter any parameters (retract amount, z-hop, etc.) as part of the command.  You would need to experiment because M600 is not enabled in my firmware so I can't debug that.
3. "Support Interface Density" and "Support Interface Pattern" are not settable per model and don't work with Support Blockers configured as modifier meshes.

The round model in the image is a poor candidate for "Support Interface Material Change" because so many pauses would be required.  The square "box top" model is (and was) perfect.  Printed by itself - I set the "Support Interface Density" to 100% using "Lines" and with the "Support Top Distance" set to "0".  The results were outstanding.
[image]

With the "Per Model" settings of the box top set to "Support Top Distance" of 0 both of these models could be printed together.  The "Layers of Interest" would be entered as "10, 90" for the box top and there would be 4 pauses in the gcode.  The "Layers of Interest" are the top layer of an Interface.  With the top layer of the "support interface" in PETG on a PLA model - the support will (does) pop right off.
[image]

You can download the post processor from [HERE](https://github.com/GregValiant/GV-PostProcessors/blob/StartPoint/scripts/SuptIntMaterialChange.py) by using the "Download Raw File" button.  Move it to your **Configuration Folder** and in the **"scripts"** sub-folder.  In Windows that would be something like _"C:\Users\...your username...\AppData\Roaming\cura\5.8\scripts"_.
Open the file in a text editor and read the "Rules" at the beginning of the file.  
I have not submitted this as a pull request as it is not a simple "install and forget" script.  You must carefully view the Cura preview to insure you have Cura set up correctly.
For a PLA model, PETG makes a good support interface and for a PETG model, PLA makes a good support interface.  TPU seems to stick to everything so this post processor probably wouldn't work well with TPU.  For other materials you would need to experiment to find which ones work with each other.
If you decide to try it I would appreciate some feedback.

Having support come off like this is what I was shooting for.  (I've been using this to mix epoxy).
[image]

jacc190676, 2024-08-03:
Thank you so mutch for all these explanations ! Yes, i forgot round elements, it's exactly how you describe.... i will have a look to the attached files and all your ideas, specially for the flats, sounds they can be usefull for me from time to time