Ultimaker/Cura #19455: Horizontal bore: Allow inward conic support

herrdeh, 2024-07-31:
### Is your feature request related to a problem?

There seems not to be a way to remove supports in front of a horizontal bore:
[according to this request.](https://community.ultimaker.com/topic/45885-st%C3%BCtzstruktur-vor-horizontaler-bohrung-entfernen/)

When creating horizontal bores, for the smaller ones you need to drill supports out, but that's difficult if it looks like this:

[image]

### Describe the solution you'd like

For this reason I'd appreciate an option for inward conic support of horizontal bores, looking like this:

[image]

### Describe alternatives you've considered

An option to remove support on front of horizontal bores can be useful, but is not as good as the solution described above.

### Affected users and/or printers

Any.

Asterchades, 2024-08-01:
You can already configure Conical Supports (under Experimental) with a negative angle. This will naturally cause it to apply to _all_ supports on a given model, but may provide the functionality you're after. For some specific examples a positive angle may also work.

As for the supports in front of bores like that, my experience has been that these are caused by **Use Towers** being enabled in the support settings. These seem to occasionally detect parts of the model which they shouldn't, causing a tower to be generated next to the model that doesn't appear to go anywhere.

herrdeh, 2024-08-12:
Thank you for your kind answer, sorry for the late reply, it took some time to find out what the feature does.

This is what I could achieve:

[image]

Indeed, the structure in front of the bore is somehow "conic". Unluckily, it's not a solution to my question. 

Trying to be more precise:
To remove support structures inside a horizontal bore, using a rotating drill seems to be the obvious solution. But the above kind of support does not help to find the centre of the bore, thus creates the danger to damage the bore's surrondings.

For that reason, I'd appreciate  a true conic, and centered-to-the-bore support, shape can be seen at 2nd image of this thread.