Ultimaker/Cura #19456: 3mf export with multiple objects doesn't work well

rondlh, 2024-07-31:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

Custom

### Reproduction steps

1. Add the attached "Original.stl" to Cura
2. With the "Mesh Tools", "Split models into parts" (Mesh Tools from Marketplace, then right click the model).
3. Export the object to 3mf format
4. Close Cura
5. Import the 3mf file

The picture below shows the original stl file and the imported 3mf file.

[image]

### Actual results

See picture above, parts seem to have moved and rotated.

### Expected results

Parts position stays the same after exporting to 3mf

### Add your .zip and screenshots here ⬇️

[image]

The original STL file (left in picture above) and the exported 3mf file:
[Original - Split Mesh, Exported to 3mf.zip](https://github.com/user-attachments/files/16439451/Original.-.Split.Mesh.Exported.to.3mf.zip)

GregValiant, 2024-07-31:
Thanks for the report.
This may be related  to #19445 .
There are two models (original stl(3) and original stl(4)) that simply will not sit flat on the build plate.  There may be something wonky in the model files of those two.  After repairing the models with MS3d Builder they sit flat, but the model orientation is still wrong once the STL is split into parts, a project file created and then re-opened in Cura.

For the other models it may also be something in the model files but at this point, I just can't tell if it's the models, Cura, or the CAD export utility.  There is no doubt that the results are wrong but why they are wrong isn't immediately apparent.
This type of problem was worked on in Cura for 5.7.2 but it doesn't appear to be fixed.  I get the same incorrect results when I create my own project file using the repaired STL's.  I used 5.8 beta1 for all my tests.
I have not had this happen with any of my own models and projects and that is one reason I think the problem could be outside Cura.  On the other hand, I can always duplicate this with other peoples project files and so I remain undecided.
So far, wherever this problem lies, it's been slippery.
When I open the attached project file in 4.13.1 it loads as we see (incorrect results).
Then I straighten out the model orientations (the two still don't want to be fixed) and create a project file using 4.13.1.
That 4.13.1 project opens correctly in both 4.13.1 and 5.8 beta1.

The Cura team will take a look.

Asterchades, 2024-07-31:
If you group the models after splitting the original STL, everything appears to work exactly as expected. They stay together, in their right places, no matter which version is used to import or export. Not intuitive, perhaps, but it works.

I'm then also able to reproduce the 4.13.1 results in newer versions of Cura - specifically 5.7 and 5.8. If after splitting the models you then use the **Reset Origin to Center** option (also part of Mesh Tools) before you export it, it will result in the stored model placement and rotation being exactly what was on screen. These can then be imported into 4.13.1 as either a model or project successfully, but must be imported into 5.7 or 5.8 as a project - importing them as models will trigger an unavoidable auto-arrange, moving and turning the pieces (not great when modifier meshes or deliberate overlaps are involved).

This is also true of a Universal Cura Project (UCP). The 3MF looks good in external tools, and imports correctly when done as a project, but triggers the auto-arrange when done as models. So this may also be an option, as these are more printer agnostic than a regular project file. 

Then of course there's the standard Export (Selection) option, which also yields a correct 3MF result but no longer has the ability to be imported as a project. As such they will import into Cura 4.13.1 correctly, but there is no way in the newer Curas to import this without the auto-arrange being triggered.

If you don't reset the origins, however, it causes problems. If you simply split and save (or export), regardless of which project type is used, the models are separated in the resulting 3MF - both horizontally and vertically, with no clear pattern. Importing these as models will trigger the same auto-arrange and, at least for me, would result in what looked like a usable build plate (nothing of mine was sunk under the surface). Much as above, though, importing as a project will maintain these incorrect locations.

rondlh, 2024-07-31:
@GregValiant Thanks for the feedback.
I was just playing around a bit to see if I could get more insight into this issue, and found the following:
1. In MS 3D Builder the 3mf files is imported correctly:

[image]

2. If I use the "Auto Orientation" extension on any part (except the bolts) then the results are not correct. So there seems to be something odd in the meshes.

Objects positioning after "Auto Orientation" (fast optimal)
[image]

@Asterchades The models seems to move after loading the 3mf file. For a very short time the location seems correct, then they "jump" to an incorrect location. Perhaps this is related to the auto-arrange you mentioned.

GregValiant, 2024-07-31:
This is making my head hurt.
Going a bit of a different way and using Mesh Tools "Reset all Model Transformations" and then creating a new project file appears to work in keeping the models where I put them.  Sort of like wiping out all the old information and putting in new.

The question remains - What is causing the models of the original project file to come into Cura at wrong locations?  It would seem that Cura should be storing and using any model transformations (location, rotation, scaling, etc.) for each mesh in the scene but I don't think that's what I'm seeing.  I wonder if that's where a Cura bug is - in the "transformation information" and the way it is stored in the 3mf file.  Once the models are divorced from their original files and become part of the project they still have bounding boxes which should have size, location, rotations.  Being able to keep them oriented by "Grouping" the models may be a pointer to that as well.

rondlh, 2024-08-01:
@GregValiant
> Going a bit of a different way and using Mesh Tools "Reset all Model Transformations" and then creating a new project file appears to work in keeping the models where I put them. Sort of like wiping out all the old information and putting in new.

This does not work for me in Cura 5.7.2. When creating a new project the parts positions are different than when using the export function. There seems to be kind of a staircase going on. It seems the loading of one object affects the next object. Looking to the bolts, 1 has an offset (to another bolt) of exactly 5mm, another one has an offset of exactly 10mm. 

[image]

Loading the 3mf in 3D builder is not correct this time.
[image]

rondlh, 2024-08-04:
Update: Cura 5.8.0 has the same issues

nallath, 2024-08-05:
I think that this PR https://github.com/Ultimaker/Cura/pull/19375 might actually fix the issue.

rondlh, 2024-08-30:
Update: Cura 5.8.1 has the same issues

rondlh, 2024-11-21:
Update: Cura 5.9 still has this issue, but behaves quite differently.
On the left the view when saving the 3MF, on the right 3MF-file loaded again.
Everything part seems to have moved to the center. I checked for settings that could have caused this, but didn't find anything.

[image]

GregValiant, 2024-11-21:
Just a data point:
If I save a project file, and elect to open it as a "project file", it seems to be OK.
If I open that project file and select "Import Models" then the parts move around.

rondlh, 2025-03-26:
I just tried saving and loading the mentioned 3MF file again in Cura 5.10.0, and it worked correctly.
Great job, thanks for the support!

closed 2025-03-26 by rondlh