Ultimaker/Cura #19457: [5.7.2] Zig-zag support pattern is adding extra lines on slanted top/bottom surfaces

WhyIcanBuyMe20atAnyMarket, 2024-07-31:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

Sindoh 3DWOX — non-standandard, added in program files

### Reproduction steps

1. Open model with a slanted surface (does not happen on level surfaces)
2. Slice with zig-zag support

### Actual results

Single lines of support, on the layers directly adjoining both top and bottom surfaces, in between where the zig-zags are in intermediate layers. These will be almost impossible to remove, and will substantially affect the size of the hole, which needed to interact with a different part.

### Expected results

Just the same shape of support as in-between. Especially on the bottom of the support, there's no point having anything else.

### Add your .zip and screenshots here ⬇️

[image]
[SideSpadeWeightTester v3.zip](https://github.com/user-attachments/files/16443017/SideSpadeWeightTester.v3.zip)

GregValiant, 2024-07-31:
Thanks for the report.
The 3mf file opens for me as just a model file rather than as a project file.  That means I'm not getting your settings and printer.
Trying to recreate this I see this at the top interface:
[image]
That "extra support line" going across is to anchor the "grid" interface and so it needs to be there.  The same line on the bottom appears (I think) because the calculations are the same regardless of whether it's for a top or a bottom.

I'm going to remove the bug label as this looks like "expected behavior" and I'll go ahead and close it as well.  Let me know if you feel that is premature.  If you post a project file and something indicates a problem then this can be re-opened.

WhyIcanBuyMe20atAnyMarket, 2024-08-01:
Sorry about the file. I've updated it now and checked that it contains the settings. There isn't any support interface. The line is a normal support line, it's just in a silly place, at the edge of the support (for that layer) but not underneath/over anything else.

WhyIcanBuyMe20atAnyMarket, 2024-08-01:
Just to clarify, I switched the .3mf file in the original post for a proper project file, hence there wasn't an attachment to the message above.

WhyIcanBuyMe20atAnyMarket, 2024-08-01:
My file settings are probably a bad example for why this is detrimental on the top, as the support line spacing is less than the skin overhang per layer, so these unexpected lines could conceivably support overhangs better. However, the same problem is present regardless of support density, as seen in the the screenshots below at 50% support density, and I see no reason that it should be on the bottom of the support regardless of density
[image]
[image]

GregValiant, 2024-08-01:
I'll re-open this for you.  Right now that's the way it is supposed to work.  I think that "additional density" just below the support interface was added in 5.7.1 as a way to provide support for the first layer of the interface.
Someone will take another look.

WhyIcanBuyMe20atAnyMarket, 2024-08-01:
Right. Shouldn't that be disableable then when there isn't an interface? Perhaps a feature request instead of a bug.

WhyIcanBuyMe20atAnyMarket, 2024-08-01:
No, I have just tested in 5.1.0 and the issue is still apparent there.
[image]