Ultimaker/Cura #19458: Not translated lines which have \n

h1data, 2024-07-31:
### Cura Version

5.8.0

### Operating System

Windows 10

### Printer

not spesific

### Reproduction steps

Open printer configuration and mouse over on text areas for G-Code.

There are `\n` in fdmprinter.def.json.pot and po file.
``` pot
msgctxt "machine_start_gcode description"
msgid "G-code commands to be executed at the very start - separated by \n."
msgstr ""
```

``` pot
msgctxt "machine_end_gcode description"
msgid "G-code commands to be executed at the very end - separated by \n."
msgstr ""
```

`\n` in pot and po file is recognized as an actual line feed. I tried escaping like `\\n` but it did not work.

### Actual results

The tooltip texts are not translated.
(see attached image)

### Expected results

Displays translated texts.

### Add your .zip and screenshots here ⬇️

[image]

wawanbreton, 2025-03-14:
Oh, good catch...  I have added a bug on the backlog (CURA-12467), and we will try to fix it before release. If we need to change the source text, we will update the according `po` files as this is just one label.

h1data, 2025-03-15:
I've tried to change `\n` to `\\n` in `msgid` before, but it didn't solve.
If it needs to fix codes, it's okay to fix this after 5.10.

wawanbreton, 2025-03-17:
So, I took a quick look and it turns out that we have removed this tooltip a while ago: https://github.com/Ultimaker/Cura/commit/5bff7067b0b28d0e17a81e2ecea8ff17d41a964a
It was more annoying than helping, and we also added an online documentation that should help more. As of today, the description for the start/end gcode is then not used. An intermediate solution woud be to set it on the title that is above the text area, but not anyone would have it shown on purpose. So I would just keep it as is, what do you think ?

h1data, 2025-03-17:
Yes, 5.10 will introduce new G-code features and
I agree that detailed documentation helps much more than tooltips.
I think it's ok to close.

closed 2025-03-17 by h1data