Ultimaker/Cura #19459: Print not change to printing temperature,

QTronics, 2024-07-31:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

VZBot

### Reproduction steps

When I set the printing temperature, Cura automatically sets the initial printing temperature 10 degrees lower. When I slide and send it to my printer it never switches over to the printing temperature. It stays at the initial printing temperature.

### Actual results

Printing temperature toggle is not added to g-code

### Expected results

Expected to change from initial printing temperature to printing temperature at some point during the print.

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/user-attachments/files/16447392/cura.log)
[image]
[TestPrint_200x200.zip](https://github.com/user-attachments/files/16447432/TestPrint_200x200.zip)

GregValiant, 2024-07-31:
Thanks for the report.
This is from your gcode file:
M104 S235  ; Added by Cura to set the hot end to 235 (your setting)
M105           ; Added by Cura to "check temperature" and send the response to a print server if you are using one.
M109 S235  ; Wait for the hot end to hit 235 before continuing
M82 ;absolute extrusion mode
PRINT_START EXTRUDER_TEMP=225 BED_TEMP=105.0  ;???? It might be in your StartUp Gcode and it lowers the temp to 225.
SET_MATERIAL MATERIAL='ABS'

So the problem appears to be in your StartUp Gcode.  You will need to post a project file so your printer information is included.
If you will post a project file I'll take a look but right now it is not a bug but rather "expected behavior" given that the problem is not with any gcode command but rather some firmware macro commands which seem to be misconfigured.  Cura did not add those commands.  I'll leave this open so you can respond.

What you should see in your startup gcode is this:
PRINT_START EXTRUDER_TEMP={**material_print_temperature_layer_0**} BED_TEMP={**material_bed_temperature_layer_0**}

My guess is that what you have is this:
PRINT_START EXTRUDER_TEMP={**material_initial_print_temperature**}  BED_TEMP={material_bed_temperature_layer_0} "material_initial_print_temperature" indicates that the hot end should use that temperature (225 in your settings).  That setting is for multi-extruder machines to use at tool change.  An alternate problem would be if the "225" is hard coded into your StartUp Gcode.

QTronics, 2024-07-31:
Thanks. I see my error. Your help is appreciated.

GregValiant, 2024-08-01:
I'll go ahead and close this.

closed 2024-08-01 by GregValiant