Ultimaker/Cura #19460: Support not generating correctly over sloped model surfaces.

jonwienke, 2024-08-01:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

Any

### Reproduction steps

Generating support Everywhere over portions of a model where the upper surface is not flat, 

### Actual results

There are gaps between the upper surface of the model and the support infill above it. This causes the support to print like garbage over sloped model surfaces because of excessive overhangs over the gaps where support interface floor is supposed to be.

### Expected results

The support interface floor should fill the space between the upper surface of the model and the support infill above the surface.

### Add your .zip and screenshots here ⬇️

[UCP_DodecahedronSphere.zip](https://github.com/user-attachments/files/16448726/UCP_DodecahedronSphere.zip)
[image]
[image]

Asterchades, 2024-08-01:
I'm unable to reproduce this with the file as provided. I can only suspect that there are settings required which are simply not being exported in a UCP, so instead a full project - which includes _all_ currently used settings - is going to be required to perform diagnostics.

jonwienke, 2024-08-01:
[NCP_DodecahedronSphere.zip](https://github.com/user-attachments/files/16450942/NCP_DodecahedronSphere.zip)
Here's the other type of Cura project file. Let me know if you need anything else.

Asterchades, 2024-08-01:
I'm definitely able to reproduce this with 5.7, 5.7.2, and 5.8 (currently in beta). Though curiously the results were different (and arguably better) immediately after importing - it wasn't until after closing and re-opening with the updated profile already in place that the results match what is shown here.

This is _possibly_ related to #19442, but I'll leave it for someone more knowledgeable on the issue to determine if it's a duplicate or a separate issue.

jonwienke, 2024-08-01:
I tried adjusting all of the support interface settings (turning support interface floor on and off, adjusting thickness, etc) and zeroed out every support distance setting I could find, and couldn't make it go away.

Asterchades, 2024-08-01:
I did try a few combinations of offset, both horizontal and vertical, without success. Zero and small clearance values seemed to yield the same basic results. Removing the minimum support areas also appeared to have no impact, nor did applying a small amount of Support Horizontal Expansion.

I suspect this may be related to multiple extruders as my single-extruder setup didn't show the same issue. Unfortunately multi-extruder stuff is well and truly outside of my knowledge base.

LilBub, 2024-08-01:
I'm had those little specs/dots occur in the past, especially when the first layer is being supported with tree supports.  Sometimes brim helps save them (and the print) but yeah, it can be weird how Cura builds that first layer.  Most of my prints do not have a flat surface so I have to re-orient the part a good deal just to get a solid support first layer.  I can't seem to find the model this happened on at the moment and the ones I have here at work don't display the spec/dots thing.  I'll search some more to support you! (See what I did there?)  :-)

jonwienke, 2024-08-01:
> I suspect this may be related to multiple extruders as my single-extruder setup didn't show the same issue. Unfortunately multi-extruder stuff is well and truly outside of my knowledge base.

I tried changing everything in the project to print from extruder 1, and the weird gaps and other artifacts persisted.

jonwienke, 2024-08-20:
> Another thing I noticed duting printing is all the little flecks of support at the corners of the support interface. The printer wastes a bunch of time hopping around dropping uselessly tiny flecks of support material instead of just printing the lines, which makes the support surface rougher and makes the extruder more likely to jam because it isn't actually extruding much of anythng while printing that mess.
> 
> It does it on roof, too.

I figure out what's happening more specifically for part of this. Normal layer thickness is set to 0.15mm, and the first layer and support infill layer thickness are set to 0.3. What's happening is that in the layers where support infill shouldn't be printed at all, Cura is printing garbage specks at both ends of every support infill line:

[image]

It's possible  this is a separate issue from the gaps in support where they shouldn't be, but there's no point telling Cura to only print support infill every other layer and have it print support infill every layer anyway.

Asterchades, 2024-08-21:
It's hard to tell if they're separate or not. Given the rest of the weirdness it's entirely possible they're related, but even changing the infill height to match the layer height doesn't seem to remedy the major problem here.

I'm going to give @nallath a ping on this one, just to make sure someone further up the chain has had eyes on this. That's not to say that they haven't already - they could have easily looked and not commented - but I just want to make sure, as I'm completely unable to reproduce this with my own settings and unable to make yours behave any better.

jonwienke, 2024-08-21:
Thanks for looking into it.

Making all the layer heights the same makes some of the random specks go away (they merge into support infill lines), but not all. The gaps and random tomfoolery around the gaps remain.

jonwienke, 2024-08-25:
One more clue:
Changing **Support Distance Priority** from **Z overrides X/Y** to **X/Y Overrides Z** seems to make the bottom gap issue go away. So that's something glitched in code that only runs when **Z overrides X/Y** is selected.

The issue with the support infill specks printing at support infill line ends on layers where support infill shouldn't print at all persists regardless of the **Support Distance Priority** setting.