Ultimaker/Cura #19462: [5.7.2] Support roof fused with model

Skydef, 2024-08-01:
### Cura Version

5.7.2

### Operating System

Windows 10

### Printer

Anycubic i3 Mega

### Reproduction steps

I've upgraded from Cura 5.5.0 to 5.7.2 and encountered a problem with a model that was working fine previously. The support roof is too close the the model structure. I've tried to remove the support after printing but even with excessive amount of force it is barely removable since it's largely fused with the model. Print profile is "Normal". All Settings are default. In both Cura versions "Support Z distance" is 0.1mm (default).
1. Load .stl file
2. Settings: Anycubic i3 Mega, eSun PLA+ Black, Profile Normal
3. Slice

(The issue can be mitigated by increasing Support Z distance e.g. to 0.2mm)

### Actual results

Cura 5.7.2. No space between support roof and model:
[image]

### Expected results

Cura 5.5.0. One layer of space between support roof and model:
[image]

### Add your .zip and screenshots here ⬇️

[ring-element-screw-connection.stl.zip](https://github.com/user-attachments/files/16457934/ring-element-screw-connection.stl.zip)

LilBub, 2024-08-01:
Can you attach your project file?  I think this is something in your settings as in 5.7.2 I am seeing the Support Interface just fine.

[image]

Skydef, 2024-08-01:
Thanks for looking into it. I've attached my project file to the issue.

Asterchades, 2024-08-02:
I'm almost certain this is due to the way that the top of supports was changed with the introduction of Cura 5.6. Specifically this note:

> Updated the Support Z Distance to allow for more different values, no longer only multiple layerheights, solves https://github.com/Ultimaker/Cura/issues/6105

In 5.5, you're getting a full layer because it's rounding up to the next full layer. It has no choice. So your specified 0.1mm offset is becoming 0.2mm, yielding a full layer gap between the two.

5.6 and onward no longer perform this rounding. Cura's preview engine isn't really showing it clearly, but the layer immediately under the model is being printed with a different Z value and correspondingly reduced flow (it will appear narrower, rather than shorter). End result is your specified 0.1mm offset actually resulting in a 0.1mm offset, and for your setup this is simply too small to print as two distinct, separable parts.

Few (if any) third party definitions have received updates making accommodations for this change, and the global default has also remained unchanged. As such if you still want a full layer you will need to manually adjust your settings.

Skydef, 2024-08-03:
This sounds like the cause of it. Perhaps a little higher default value should be set then to accommodate for the changes? If you think the current default value is still fine, then the issue can be marked as solved.