Ultimaker/Cura #19471: [5.3.1] Part of thin wall missing

ymh1003, 2024-08-03:
### Cura Version

5.3.1

### Operating System

MacOS Monterey

### Printer

Voron Trident 300

### Reproduction steps

Open the project files and slice the models.

### Actual results

Part of the model is missing in only one of the model, although all of the three models have the same thickness (0.2mm) in their thinnest part.

### Expected results

They should slice in the same pattern.

### Add your .zip and screenshots here ⬇️

[project.zip](https://github.com/user-attachments/files/16482597/project.zip)

The zip file contains three 3mf files. The one called `wall-thickness` is the model which is sliced incorrectly. 

[image]

I created this model based on an existing model which is also sliced wrong in a similar pattern. The thinnest part in this model has a 0.2mm thickness, so I created the other two models to test whether it is caused by the thickness. However, the other two models slice correctly though they all share the same thickness at the thinnest part. So I am confused what causes this issue.

Asterchades, 2024-08-04:
Try it again in 5.8 (released three days ago). There were fixes implemented which specifically address small areas like that failing to slice, which seem to work perfectly on this both as a model and a project.

ymh1003, 2024-08-05:
This model still won't slice correctly in 5.8, as part of the thin wall is missing:  
[VT300_ERCF_part.3mf.zip](https://github.com/user-attachments/files/16490275/VT300_ERCF_part.3mf.zip)

Moreover, I have modified the models I attached above to have the same thickness 0.1mm at their respective thinnest part. As in 5.3.1, this model still slices wrong, while the other two slice correctly.
[image]
[image]
[image]

This zip file contains the above models: 
[0.1mm.zip](https://github.com/user-attachments/files/16490347/0.1mm.zip)

Also, in 5.8, when I put these three models together on the build plate, the sliced results for two of the models (left&right) are different compared to the sliced result when they are the only model on the plate.
[image]

This behavior doesn't seem to appear in 5.3.1.
[image]