Ultimaker/Cura #19475: filament run-out sensor not triggering

goofyseeker311, 2024-08-04:
### Cura Version

5.8.0

### Operating System

Win11 RTX3080

### Printer

elegoo neptune 3 pro

### Reproduction steps

1. print something
2. run out of filament while printing

### Actual results

printer continues printing while it should have stopped when filament sensor triggered run-out

### Expected results

normally when printing from sd card the printer stops itself when it detect filament run-out. when printing to usb from cura, the filament run-out sensor is ignored. dont know if this is a feature or a bug.

### Add your .zip and screenshots here ⬇️

nothing to add to the filament sensor.

GregValiant, 2024-08-04:
Thanks for the report.
The only feedback from the printer to the USB Printing plugin in Cura is responses to temperature queries (M105).  Cura uses that to make sure the connection is still alive.
It would seem to me that the printer should be able to tell if the runout sensor has been tripped.  If your mainboard is ignoring the sensor when the gcode is coming into the USB port - that would seem to be a firmware problem rather than a gcode problem so I'm going to remove the bug label from this.  Let us know if you feel that is premature.

USB Printing is kept around for legacy reasons as some of the original UltiMaker printers require it.  It is not maintained and is subject to hardware incompatibilities and other buggy behavior.  It is always recommended to use the SD card.

goofyseeker311, 2024-08-05:
Well I have decently recent firmware on the 3d printer, and the printer just continued to print without filament, happily.

GregValiant, 2024-08-05:
_"decently recent firmware"_
Did you change the firmware recently?
When the runout sensor is tripped, the firmware monitoring the runout sensor "pin" on the mainboard has to notice that the pin has changed states (from open to closed or vice versa).  The firmware must be configured to understand the change in the pin state and do what it needs to do to stop the print.  That might mean sending an M0 or M25 to itself.  If the firmware is also configured to send a signal back to a "print server" (like Octoprint or Repetier Host) then maybe the flow of gcode would stop.  The Cura USB Printing plugin isn't that sophisticated and would just keep sending gcode even through the printer had stopped because the filament runout sensor was tripped.
The runout sensor is on the printer side of things and therefore everything needs to be handled on that side.  It has nothing to do with Cura at all.

Ultimaker determined years ago (2017?) that USB Printing was glitchy and prone to errors.  That's why it was abandoned in favor of network printing.  Anyone who could fix the USB Printing plugin would be a hero.  No one has take up the challenge because there are thousands of combinations of operating systems, firmware, mainboards, printer/processors, USB port hardware and configurations and port drivers.  There are still a lot of "Ultimaker Originals" in service around the world and they require USB printing as it is the only way they work.  As a consequence, USB Printing is optimized for that particular hardware and firmware.  Any other combination becomes a crap shoot was to whether it works or not.

I bought a "[female SD to male micro SD](https://www.amazon.com/LANMU-Extension-Flexible-Monoprice-Raspberry/dp/B07YHN83NJ/ref=sr_1_3?crid=1T31LQ0NAJ2U4&dib=eyJ2IjoiMSJ9.ZPPYU-mO5oJBZMhoKSwS6XyGvc76RpPWscOks_z3qHp0ZjbVSG_pb6pVtLobxSnBU2KvAm09Nv8LHwA2l6Fv6Vp_t9zA9bUyfskxD83wQ1_LhmhD1VEo7_WYf2r7Pl29V08iAdiH_m-HaaqecU0tJj9PpcjHU6TWoEtW0bq9MpjbIhZTfCeK88p2khmPNTj7PHArSzB9ZBcFC4jK5bR7O68roUrjWK8eK_pVed1VS7I.Djg0f1U90o4kZKKnnBCRUmzZGAHEkKjyFYdOp-mQi3c&dib_tag=se&keywords=female%2Bsd%2Bto%2Bmale%2Bmicro%2Bsd%2Badapter&qid=1722869701&sprefix=female%2Bsd%2Bto%2Bmale%2Bmicro%2Bsd%2Badapter%2Caps%2C194&sr=8-3&th=1)" converter.  I designed a little box that bolts to the front left leg of my Ender and I leave the converter plugged into the printer.  That allows me to use the larger SD card and there isn't any wear and tear on the little socket on the mainboard.

I'm going to go ahead and close this.  It just is not a Cura problem.