Ultimaker/Cura #19478: Cura 5.8.0 Infill problem

Schatta123, 2024-08-04:
### Cura Version

5.8.0

### Operating System

Windows 11

### Printer

Biqu B1

### Reproduction steps

1. I created an object that has a thickness of 3mm
2. I slice the model with a 30% cubic infill

### Actual results

Cura wasn't able to put infill. It just left it completely hollow.
[image]
 

### Expected results

While in cura 5.7.0 it is able to put infill there like I expected. I even use the same profile and object.
[image]

### Add your .zip and screenshots here ⬇️

[3mm Problem.zip](https://github.com/user-attachments/files/16488652/3mm.Problem.zip)

GregValiant, 2024-08-04:
Thanks for the report.
I can duplicate this.  With 3 walls at 0.40 line width, the walls take up 2.4mm of the model.  That leaves just a 0.60mm gap in the interior.  Since that is less than 2 line widths wide there is no infill.
If you decrease the Wall Count to 2 you will see that infill is generated.
If you decrease the Wall Line Width to 0.37 then you will also get infill.
Another alternative is to make the Wall Count 4 and make the piece solid.  It's so small that infill won't have any effect anyway.
This is 3 walls at a line width of 0.37.
[image]

The Cura team will take a look.

mzhu1113, 2024-10-22:
Since I can't see the status of the internal ticket, is there movement on this? This is a rather frustrating issue as an ODM having to work around this for designs that are not in our control so we're left with the option of giving the customer a substandard part (not viable) or expending much more material/time (not ideal). This is provided that a slicer tech actually catches that a design would have this issue.

I do agree the idea of there being a filter but I'd much rather have preferred this was a value the user is able to modify and what to do if it's under that value though I'd imagine there's a conflict with Minimum Infill Area?

Thanks.

mzhu1113, 2024-10-23:
@HellAholic 
Thanks for the reply and also thanks for the work put into the 5.9.0 beta. Had a couple parts printed and the seams are pretty mint. 😄 

> Just as a curiosity point and maybe some additional bump up points for this, how is the final part quality affected if you go with 2 wall line count instead of 3? I'm using the initial model you shared as an example here, so it might not work in all scenarios (if you can provide additional cases that also helps)

I'm didn't submit this bug haha. But I am tracking it because we're affected by this and it's a painful one for us at that which I'll explain more below.

I don't know of why or what was the reason the filter was introduced (I assume it was a overall speed increase for the job or other problem) but I'd much rather the part is not lack internal infill like this for an arbitrary amount like "2 wall perimeters" and simultaneously sacrifice part strength. If something like this has to be made, I'd rather it be not hardcoded and something user configurable. 

With 2 wall lines and a gap present, the part loses too much shear resistence and the resolved part is able to have it's walls compressed from just simply pinching the affected features. However, all bumping the wall line count from 2 to 3 does is just change the thickness of a feature that will be affected. With 0.4mm nozzles, a 2.0mm thick panel will have this issue. Great 3 wall perimeters will fix it but now I'll have a problem with a 2.8mm thick panel and so on and some models are fairly complex and at some point simply increasing the wall line count is not going to help.

Other points I want to make
1. You have a user base that most likely would not know that this is an issue. Not everyone just looks through a github repo and navigate/search (much less know what they're looking for). So this filter is creating parts that they are going to perceive is weaker in Cura than other slicers.

2. Even *if* we know of the workaround, we would also have to know that a specific model would be impacted in order to actually change that setting. This requires human checks and with our daily throughput some part is going to be missed which leads to us having to deal with that fallout if it makes it to a client.

3. Some models prevent us from using that workaround and the workaround to that workaround such as infinite bottom/top layers actually cause undesired toolpath movements, oddly applied speed logic with bridge settings, and just plain limitations from the client.
We have a client that uses us to print full leg prothetics that has a base with holes for bolts to go thru. We have access to a Lumafield and found that we cannot go above some amount of wall line count otherwise the walls around the holes and the base itself will introduce cavities at the junctions that they have to meet. I also have infill set to 100% concentric. The problem is that some prothetics have to be thicker and the part has to be 100% solid and the base top/bottom pattern has to be rectlinear. So because we're aware of the bug/quirk we have to sink a lot of time looking for this issue if it's present in a part and it's frustrating when one makes it through after expending rather expensive material just to have to scrap and reprint when we find a small internal cavity in the CT scan where it is thicker than the 9-10 wall count and the infill won't kick in because their modeler targets a certain range of thickness.

Hope this helps. I can't really provide example models cause of some NDA or HIPPA so I'll try to describe as much as possible. I can CAD some example pieces if you need but it'd take time.

Thanks again!

GregValiant, 2024-10-23:
This is a wedge shape.  I think it shows the problem regardless of wall count.
[Infill Test.zip](https://github.com/user-attachments/files/17498108/Infill.Test.zip)

smartavionics, 2024-10-23:
Hi @GregValiant , for amusement, I sliced your test project with no alterations using 4.20.24 and this is what you get...
[image]
[image]

Pretty well filled with either infill or gap-fill.

So folks, if you don't need Cura 5 features, there is a solution at https://github.com/smartavionics/Cura/releases

rburema, 2024-10-24:
For those who might want to follow along, the fix is implemented here; https://github.com/Ultimaker/CuraEngine/pull/2156 and indeed slated for 5.9.0-final once merged (not in the 5.9-beta, it's still broken there).

mzhu1113, 2024-10-24:
Perfect! Thank you everyone for pushing this through to the stable release! I'm really looking forward to this one.

rburema, 2024-11-07:
FYI: Due to unrelated circumstances, we're decided to have a 2nd beta this time around. The upshot is that this already should include the fix: https://github.com/Ultimaker/Cura/releases/tag/5.9.0-beta.2 (and of course it'll still be in stable later on).