Ultimaker/Cura #19479: wont slice and says bug

Jerzygraphics, 2024-08-04:
### Cura Version

5.6

### Operating System

windows 11

### Printer

ender 3 v3 ke

### Name abnormal settings

no

### Describe model location

yes

### Describe your model

no

### Add your .zip here ⬇️

[deadpool_mask_main_cover_head_four_pieces_onirigena_by_eikki.zip](https://github.com/user-attachments/files/16489171/deadpool_mask_main_cover_head_four_pieces_onirigena_by_eikki.zip)

Jerzygraphics, 2024-08-04:
wont slice and says bug detected

GregValiant, 2024-08-04:
Thanks for the report.
Cura 5.5 and 5.6 had problems slicing certain shapes and "organic" rounded shapes were a particular problem.  Cura 5.8 was released a few days ago and is much better so I suggest it's time to upgrade.
I'll mark this as a duplicate and go ahead and close it since 5.6 is 4 versions behind.  I hope you understand.

closed 2024-08-04 by GregValiant