Ultimaker/Cura #19485: Simple part slicing error

Zebulon67fr, 2024-08-05:
### Cura Version

5.8.0

### Operating System

Linux Mint

### Printer

PROFORGE 4

### Describe model location

Model very simple in center of the plate

### Describe your model

This is a rectangular cover box

### Add your .zip here ⬇️

[P4_3D Printed Screen Case.STL.zip](https://github.com/user-attachments/files/16496351/P4_3D.Printed.Screen.Case.STL.zip)

GregValiant, 2024-08-05:
Thanks for the report.
I didn't have any trouble slicing the STL.  We'll need a Project File ("File | Save Project") to investigate this.

Zebulon67fr, 2024-08-05:
Hi
This problem does not happen in Windows 11 version

Zebulon67fr, 2024-08-05:
Here is the project file under linux Mint (Ubuntu)
Xavier

GregValiant, 2024-08-05:
You must post directly to the site.  Email does not work (Github rules, not mine).