Ultimaker/Cura #19488: printhead takes travel path during combing which doesn't exist in travel path preview, resulting in thick stringing

keishii3d, 2024-08-06:
### Cura Version

5.7.2

### Operating System

win 10 gtx 1080

### Printer

sovol sv06

### Reproduction steps

repost of my forum post:

In an attempt to root out the cause of thick stringing, I've been going nuts trying to change settings, restoring defaults, googling, etc.

It wasn't until I did a test piece, stopping it after ~20 layers, that I noticed something. The strings on the print follow a path which is non-existent in my cura preview. It appears that ~50% of the layers follow the correct combing travel path, and the other half ignore it, moving straight from the end point of one layer, to the start of the next.

I finally feel less crazy after my journey down a rabbit-hole trying to fix an issue which I believed was caused by my tinkering with combing. I assume I could get rid of the stringing by changing my combing distance before retraction from 0 to 10 or something. But at the same time, I see the printer taking a path that does not exist in the slicer preview, and it makes me wonder if I've come across a bug or something, and thought I'd post a report

 Attached are: photos of the test print "bottomless", the slicer preview, and the print with a red line highlighting the occasional correct path being used. I've also attached the project file as well as the gcode, which I've taken from the SD card after the print, to make sure it's the exact file used for the print. 

### Actual results

Since my issue results from the printer following a travel path that doesn't exist in the slicer preview, I haven't done anything else.

### Expected results

Since my issue results from the printer following a travel path that doesn't exist in the slicer preview, I haven't done anything else.

### Add your .zip and screenshots here ⬇️

[print not matching gcode travel in cura.zip](https://github.com/user-attachments/files/16503687/print.not.matching.gcode.travel.in.cura.zip)

keishii3d, 2024-08-06:
From a forum-post suggestion, I ran a test after disabling fluid motion in cura, resulting in no change.
[image]

I then closed the project without saving, and reopened it to reset the changes from the last test. I set "Max Comb Distance With No Retract" from 0mm to 10mm, and the stringing vanished as expected.
[image]

I'll use this to solve my issue for the printed parts I need, which is nice. But the travel deviations still puzzle me.

Despite finding a "fix" for my issue, I'll leave this ticket up due to the printed travel deviations from the cura preview

LilBub, 2024-08-06:
That's very interesting, I did a print last night that had some unusual stringing in 5.8.  I'll examine my settings and see if I align with your observations