Ultimaker/Cura #19489: CureEngine build error.

HonnyChen, 2024-08-06:
### Cura Version

5.8.0

### Operating System

Windows 11

### Printer

None

### Reproduction steps

Running Cura from Source and Installation step 3.

### Actual results

Has build error.
[image]

### Expected results

Build success.

### Add your .zip and screenshots here ⬇️

None.

HonnyChen, 2024-08-06:
It look like  a issue to special words. I changed another words and it's fixed.
[image]

runruiliu, 2024-08-13:
Please tell me which folder of the process.cpp file you modified. I tried to modify  the C:\.conan\2c0d14\1\src\progress file, but when I executed the step 3 command again, the file was overwritten by the previous file, and this error was still displayed.
[image]

runruiliu, 2024-08-13:
I had the same problem.Please tell me which folder of the process.cpp file you modified. I tried to modify the C:.conan\2c0d14\1\src\progress file, but when I executed the step 3 command again, the file was overwritten by the previous file, and this error was still displayed.
[image]