Ultimaker/Cura #19490: Object clashing on build plate

alavabeer, 2024-08-06:
### Cura Version

5.8.0

### Operating System

Windows 10

### Printer

Artillery Sidewinder X3 Pro

### Reproduction steps

1. Load object onto build plate
2. Press Ctrl + m
3. change the value to plus 3
4. press ok

### Actual results

Two out of the four object are clashing

### Expected results

All four object should be in their own place with space around them.

This did not happen in 5.0.0

Tried to add the zip file and got the following message:-

We don’t support that file type.

Try again with GIF, JPEG, JPG, MOV, MP4, PNG, SVG, WEBM, CPUPROFILE, CSV, DMP, DOCX, FODG, FODP, FODS, FODT, GZ, JSON, JSONC, LOG, MD, ODF, ODG, ODP, ODS, ODT, PATCH, PDF, PPTX, TGZ, TXT, XLS, XLSX or ZIP.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-08-06:
Thanks for the report.
There are additional tools in the "right-click" menu.  They were split off from "Multiply Models" for added flexibility.  "Multiply Models (CTRL+m) now just adds copies of a model to the build plate.  That is expected behavior.

Right-click on the build plate and try "Arrange all Models" and "Arrange all models in a grid" and see how they arrange the models.
Let us know if that works for you.

alavabeer, 2024-08-06:
Hi Greg,

Thanks for your speedy response and the info.

Both of the options worked. 🙂

Kind Regards,

Steve

GregValiant, 2024-08-06:
I'll go ahead and close this then.

closed 2024-08-06 by GregValiant

alavabeer, 2024-08-07:
Thanks Greg. 🙂

Closed #19490<https://github.com/Ultimaker/Cura/issues/19490> as completed.

alavabeer, 2024-08-12:
Hi Greg,

I don't know if you can help with this but I thought that I'd ask anyway.

I'm using Cura 5.8.0 and I have noticed that there are no printer profiles for any of Artillery's new machines.

I have an X3 Pro but I can't seem to find a way to create my own profile for it.

Do you have a profile available or is there a way of setting up an X3 Pro look alike?

Thanks in advance for your time.

Kind Regards,

Steve

GregValiant, 2024-08-12:
There was a pull request submitted in January ( #17965 ) that included the new Artillery printers including the X3 Pro.  Changes were requested by the automated checker, but it appears that they weren't completed by the author.  Those sorts of automated change requests are usually an attempt to make the definition file smaller by eliminating duplicate settings (a setting that changed something like "Generate Support" from "true" to "true" would throw up a flag).

The files below are from that pull request.  Use the "Help | Show Configuration Folder" command in Cura to locate your Configuration Folder.

Unzip this definition file and copy it to your "Configuration Folder" and the "definitions" sub-folder.
[artillery_sidewinder_x3_pro.def.zip](https://github.com/user-attachments/files/16581428/artillery_sidewinder_x3_pro.def.zip)

These are the nozzle files.  Unzip and copy the files to the Configuration Folder "variants" sub-folder.
[Nozzle Variants.zip](https://github.com/user-attachments/files/16581454/Nozzle.Variants.zip)

Those files have not been properly approved but I was able to install the printer and slice things.

I see in the StartUp Gcode that there is a "G28 Z" which is odd.  Maybe the "G12 C2" command calls for homing the X and Y.
In regards to the "G12" command, it is the Marlin "Clean Nozzle" command but the Marlin website doesn't list a "C" parameter.  You will need to figure out what works for your printer.  I don't own one and there is no way for me to tell what might be OK from what might be Wonky.

alavabeer, 2024-08-12:
Hi Greg,

Awesome! Thanks very much.

Will give these a go.

Kind Regards,

Steve