Ultimaker/Cura #19501: Version 5.8 crashes every time slicing is attempted

johnd-cyber, 2024-08-07:
### Cura Version

5.8.0

### Operating System

macOS Big Sur

### Printer

Creality Ender 3 S1 Pro

### Describe model location

No

### Describe your model

Model sliced OK using the previous version of Cura

### Add your .zip here ⬇️

[slicer crash project.3mf.zip](https://github.com/user-attachments/files/16521951/slicer.crash.project.3mf.zip)

johnd-cyber, 2024-08-07:
Added log file
[cura.log.zip](https://github.com/user-attachments/files/16521998/cura.log.zip)

johnd-cyber, 2024-08-07:
Discovered that the extension CuraEngine Tiled Infill Generation was causing the crash. Removing this solved the problem.

GregValiant, 2024-08-07:
This is from your log file:

2024-08-07 20:09:00,508 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [1038]: Backend exited abnormally with return code 5!
2024-08-07 20:09:00,511 - DEBUG - [MainThread] cura.BackendPlugin.stop [130]: EnginePlugin: **CuraEngineTiledInfill was killed.** Received return code -6
2024-08-07 20:09:00,511 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [421]: Attempting to kill the engine process
2024-08-07 20:09:00,512 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [427]: Killing engine process
2024-08-07 20:09:00,512 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [430]: Engine process is killed. Received return code 5

@hellaholic of the Cura Team will take a look.

johnd-cyber, 2024-08-07:
Thanks. I can confirm that removing that extension restores the slicing functionality.

MariMakes, 2024-08-13:
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-12091

Thanks for your report. Together we can make Cura better 😊

wawanbreton, 2024-08-16:
Hi @johnd-cyber,
I tried to reproduce your issue with different ways, but nothing made the plugin crash for me :thinking: 
Could you please send the plugin log file ? According to the Cura log you provided, it should be at `/Users/johnd/Library/Application Support/cura/5.8/CuraEngineTiledInfill.log`. Hopefully that will give me more information about what is happening.
And also, did you ever get the plugin to work and it stopped working at some point, or did it always crash ?

johnd-cyber, 2024-08-17:
Hi
Attached is the log file as requested. However, I suspect it won't be of much help as I wiped this directory tree when trying to resolve the problem.
I hadn't had a chance to use the plugin, but originally slicing worked, and the problem only occurred when I updated to Cura 5.8
[CuraEngineTiledInfill.log](https://github.com/user-attachments/files/16644651/CuraEngineTiledInfill.log)

Triffen, 2024-08-18:
Yes, Tiled Infill plugin isn't in the Marketplace any more for some reason.

wawanbreton, 2024-08-19:
> Yes, Tiled Infill plugin isn't in the Marketplace any more for some reason.

Indeed, we unpublished it for the moment because we had several reports of crashes, and it can be very annoying for people who don't know how to dig in and make the link with the plugin. For them, Cura just becomes unusable.

wawanbreton, 2024-08-19:
@johnd-cyber Thanks for the log file, it is actually quite useful. Apparently the plugin executable can't be started because of a missing referenced function. Can you tell me what MacOS version you are running ? We build on MacOS 12, so if you have an older version it may not work.

johnd-cyber, 2024-08-19:
OK. It looks like that might be the problem. I run it on two Macs.
The one I have a problem with is running 11.7.10 (the latest version supported on that hardware).

The other Mac does not have the problem, and is running 14.6.1 (the latest version)

GregValiant, 2026-09-28:
Tiled infill was removed a couple of versions ago (5.10.0?).  I'll close this.

closed 2026-09-28 by GregValiant