Ultimaker/Cura #19504: A model is sliced with non printable wall even if model's line width is more or equal than line with in slicing parameters

bagrusss, 2024-08-07:
### Cura Version

5.8.0

### Operating System

MacOS 13.6.4

### Printer

FlyingBear Ghost 5

### Reproduction steps

Just use attached model and profile from zip file and slice it.
1. The model should have a wall with 0,4-0,5 mm thickness, for instance.
2. "Wall Thickness" parameter is installed to 0,8 or 1,2 mm.
3. Slice the model
4. See the "Actual results" and Expected results
5. Change "Wall Thickness" parameter to 0,4
6. See the "Actual results" and Expected results

### Actual results

Step 4 - The model's circuit is printable, but is sliced with omitted part of wall on place where it's thickness is 0,4-0,5 mm
[image]

Step 6 - There are no any gaps in model's wall

### Expected results

Step 4 - The whole models wall is printable without omitted peaces (an example with Orca Slicer) - [image]

Step 6 - There are no any gaps as it was expected, but I have to use "Wall Thickness" parameter from 0,8 to get stronger printed item 

### Add your .zip and screenshots here ⬇️

[Archive.zip](https://github.com/user-attachments/files/16534794/Archive.zip)
Wall thickness 0,4 mm - no gaps [image]
0,8 mm
Wall thickness - there are two gaps in the result 
[image]

GregValiant, 2024-08-08:
Thanks for the report.
When I use the settings in the project file with 2 walls I get the same results that you did.
When I change to the stock "Dynamic Quality" with 2 walls it slices as it should.
[image]

So it appears that your settings are in a combination that doesn't work for this model.
You have "Bridging" enabled and that isn't required for this model.
You have "Outer Wall Inset" at 0.3mm and that appears to be the main problem.  That setting coupled with "Horizontal Expansion" of 0.1mm is creating a problem in that thin wall area.
A Minimum Line Width of .15 on a 0.4 nozzle will barely print and there will be no "squish".
The "Skin Expand Distance" of 0.8 will cause gaps between the skin that the upper wall is resting on, and the lower walls.

Making some changes to your settings (in particular Outer Wall Inset = 0), the slice looks like this:
[image]

So I don't see a bug here.  It appears that Cura is slicing the model exactly how you have it set up to slice.

bagrusss, 2024-08-08:
@GregValiant thanks for help! Outer Wall Inset = 0 is solved the problem. I suggest to add more tips to the described params in future releases.

GregValiant, 2024-08-09:
There are a lot more help balloons I'd like to see.  The problem is that the "ambidextrous" QT controls that Cura uses for Windows, MAC, Linux, and Chrome truly sucks at messages.

closed 2024-08-09 by GregValiant