Ultimaker/Cura #19505: slice preview

Solotov1muhammad, 2024-08-08:
### Cura Version

5.8.0

### Operating System

windows 11

### Printer

ender3 v2 neo

### Reproduction steps

1. when i do slice i can't se preview video .

### Actual results

1. when i do slice i can't se preview video .

### Expected results

the preview play button disappeared 

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-08-08:
Thanks for the report.
The "Preview Video" is not available when Cura is in "Compatibility Mode".  If you have an OpenGL version lower than 4.1 Cura will automatically put itself into "Compatibility Mode".

You can try going to "Preferences | Configure Cura" and on the "General" tab uncheck "Force Layer View Compatibility Mode" and in some cases that would work.  If Cura cannot access the graphics system OpenGL or the system doesn't support OpenGL 4.1 then Cura will stay in Compatibility Mode regardless of the value of that setting.
Cura will not work at all if the OpenGL version is less than 2.2(?).
[image]

If you cannot get Cura out of Compatibility Mode you will have to take a look at your graphics drivers.  Some systems have both a "Standard" graphic system (often an onboard Intel system) and an optional (more advanced) graphic system.  You may have to go into the graphics controller and set it up so Cura can use the more advanced graphics card.

Right now I can't tell if this is a bug, or expected behavior due to your hardware.

Solotov1muhammad, 2024-08-08:
Thank you its work

GregValiant, 2024-08-08:
Good.  I'll go ahead and close this then.

closed 2024-08-08 by GregValiant