Ultimaker/Cura #19507: интерфейс программы

st0p0r, 2024-08-08:
### Cura Version

5.8

### Operating System

Windows 11

### Printer

BIQU B1

### Reproduction steps

просто открыл программу

### Actual results

много визуальных ошибок в интерфейсе программы на русском языке

### Expected results

не должно быть этих ошибок т.к. я уже сообщал о них и вы сказали что исправили

### Add your .zip and screenshots here ⬇️

[image]

[image]

WhyIcanBuyMe20atAnyMarket, 2024-08-08:
link to previous issue
[https://github.com/Ultimaker/Cura/issues/16221](url)
[https://github.com/Ultimaker/Cura/pull/16270](url)

st0p0r, 2024-08-10:
эти линки невозможно открыть