Ultimaker/Cura #19510: New z-seam settings not available.

thre3eye, 2024-08-08:
### Cura Version

5.8.0

### Operating System

W11

### Printer

Ender 3v2neo customized

### Reproduction steps

Upgrade to Cura 5.8 from 5.7.

### Actual results

New z-seam settings aren't available. Here is a screenshot of available options:

[image]

### Expected results

New z-seam options should be available, since it seems to be the big deal about 5.8 :)

### Add your .zip and screenshots here ⬇️

[image]

thre3eye, 2024-08-08:
Ok, looks like it coms down to taking release notes too literally as a 'manual'.

I now found 'Z-Seam On Vertex' with pref search. It's an extra checkbox instead of an item in the dropdown. The release notes hint at a dropdown items and vertex being mutually exclusive. A UI bug?

"Seam Overhang Angle" is actually "Seam Overhang Wall Angle" in Experimental settings - a. can't find it by copy pasting the announcement into pref search, and b. release notes don't indicate this is experimental.

"Support Z Seam Away From Model" - alight, it's in the 'Support' section. I missed it thinking it should be grouped with other seam settings in 'Walls'

GregValiant, 2026-09-28:
I'll close this as stale.  "Paint on Supports" are now possible in Cura.

closed 2026-09-28 by GregValiant