Ultimaker/Cura #19511: 3D Viewer for preparing / preview model does not load

tferronatto, 2024-08-08:
### Cura Version

5.8.0

### Operating System

Linux (Garuda Linux OS - 6.10.3-zen1-2-zen)

### Printer

Creality Ender-3 V3 SE

### Reproduction steps

1. Open Cura
2. Observe that 3D view of the plate is blank
3. Load a .stl file
4. Observe that object is loaded and prepare options (move, rotate, etc) appear

### Actual results

The 3D view with the plate is never shown

### Expected results

I should be able to see and manipulate a model position, rotation, etc, within the plate area, as well preview the sliced model

### Add your .zip and screenshots here ⬇️

[Arquivo compactado.zip](https://github.com/user-attachments/files/16553046/Arquivo.compactado.zip)

GregValiant, 2024-08-09:
Thanks for the report.
I can't duplicate this and I have never had this problem (Windows 10 Pro laptop).
[image]

This is the first report I've seen of this behavior as well.  Someone from the Cura team will take a look.
In the meantime have you tried re-installing 5.8.0?

tferronatto, 2024-08-09:
Hi there!

I tried to reinstall 5.8.0 using the appimage from Ultimaker's page, but no
success as well. The same behavior happens.

These are my hardware settings , I forgot to include in the bug report:

OS: Garuda Linux x86_64
Host: B360M-DS3H
Kernel: 6.10.3-zen1-2-zen
Resolution: 1920x1080
DE: KDE Plasma 6.1.3
CPU: Intel(R) Core(TM) i7-8700 (12) @ 4,60 GHz
GPU: NVIDIA GeForce GTX 1060 6GB [Discrete]
Memory: 2868 MiB / 15922 MiB

TonyTwoStep, 2024-08-27:
Same issue for me on Linux (OpenSuse Tumbleweed) using the appimage.